FreeCAD/FreeCAD #13124: Standard views --> Fit all should have a new icon

FEA-eng, 2024-03-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There was a forum discussion about changing this icon (it's unintuitive now) and others agreed with it but we couldn't decide which new design to use: https://forum.freecad.org/viewtopic.php?t=83097

I wouldn't like this idea to be lost on the forum so I'm creating an issue for it. Maybe we will reach some consensus here. I can post the best designs from that forum thread and it would be nice to discuss them or look for better designs.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36391 (Git)
Build type: Release
Branch: main
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-03-24:
Here's my newest proposal:

[image]

@maxwxyz What do you think?

maxwxyz, 2024-03-24:
@FEA-eng I've also seen the forum post. I like the idea of the magnifying glass and that it should be consistent with fit selection.

FEA-eng, 2024-03-24:
> @FEA-eng I've also seen the forum post. I like the idea of the magnifying glass and that it should be consistent with fit selection.

Like this?

[image]

maxwxyz, 2024-03-25:
I like it, post it to the forum post, there seems to be a lot of opinions

FEA-eng, 2024-03-25:
> there seems to be a lot of opinions

That's why maybe it would be better to create a poll with a few proposed designs on the forum. Otherwise, the topic may die again due to endless discussions.

FEA-eng, 2024-06-13:
Closing as completed by #14180 (and then updated by #14434).

closed 2024-06-13 by FEA-eng