FreeCAD/FreeCAD #13125: FEM thermal boundary condition and loads icons should be improved

FEA-eng, 2024-03-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Icons of [thermal boundary conditions and loads](https://wiki.freecad.org/FEM_Workbench#Thermal_boundary_conditions_and_loads) are IMO too similar and unintuitive.

Let's discuss them one by one:

1. [Initial temperature](https://wiki.freecad.org/FEM_ConstraintInitialTemperature)

[image]

It's just a mesh object (used also for other FEM icons so that's ok) with a red outline. No indication that it's a temperature and an initial one (there could be a "T0" symbol or "i" letter as in the case of [Fluid boundary conditions](https://wiki.freecad.org/FEM_Workbench#Fluid_boundary_conditions)).

2. [Heat flux load](https://wiki.freecad.org/FEM_ConstraintHeatflux)

[image]

It shows that it's applied to surfaces and related to thermal analyses but it's too similar to the other icons. The current logic is that a thermometer means thermal analysis features (all have it apart from the initial temperature) while the wave symbols represent flux. I don't mind it but the icons are all too similar. Maybe this one could stay as it is and we would change the rest (as explained below).

3. [Temperature boundary condition](https://wiki.freecad.org/FEM_ConstraintTemperature)

[image]

This one is also not bad - it has the thermometer symbol and no wave symbol. It's applied to faces so only the face of the mesh object is highlighted. However,  this tool is used not only for temperature boundary conditions but also has a concentrated flux mode. I guess that it's not possible to show both modes using a single icon but we'll see what can be done with this icon.

4. [Body heat source](https://wiki.freecad.org/FEM_ConstraintBodyHeatSource)

[image]

It's too similar to the previous icons. It has the thermometer symbol and the wave symbol as a type of flux but the only difference between this one and the heat flux load is that the whole mesh object is highlighted here. I would remove the flux symbol and show some kind of heat coming out from the mesh object since that's what the tool does. It's a volume heat source. Let's just avoid making it look like surface radiation since that's another feature (not yet implemented).

Regarding the thermometer symbol, it should be either used for all thermal features with no exception (so for the initial temperature too) or removed and used as a way to represent features directly related to temperatures.

@maxwxyz FYI

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36391 (Git)
Build type: Release
Branch: main
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-24:
@FEA-eng would this be ok:
[image]

FEA-eng, 2024-03-24:
> @FEA-eng would this be ok: 

Yes, it's much better, thanks. If possible, I would just move "i" slightly to the right so that it doesn't intersect the mesh object.

maxwxyz, 2024-03-24:
@FEA-eng done. please check PR

closed 2024-04-01 by sliptonic