FreeCAD/FreeCAD #13128: Feature request: pad/pocket with fitting tolerance (same as RealThunder branch)

eblanshey, 2024-03-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

One of my favorite features in RealThunder's branch is ability to pad/pocket with a tolerance provided. For example, if a 10mm diameter hole is sketched, when pocketing it I can provide a tolerance of 0.4mm, which will make the hole 10.4mm in diameter. This is very useful because I can reuse the same sketches for multiple bodies that need to fit into each other. I want to design the sketches according to the "ideal" values, but the part's design needs to be adjusted so that the pieces can fit each other, especially if the parts are to be 3D printed.

Here's an image of the options available during a Pocket operation:

[image]

It would be great to port this to the main FreeCAD branch. I am posting this based on the [recommendation on the forum](https://forum.freecad.org/viewtopic.php?p=711754#p711754).

### Full version info

```shell
Useful version info from the RealThunder branch, since that's where this is taken from:

[code]
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 2024.322.0.14555 (Git shallow) AppImage
Build type: Release
Branch: (HEAD detached at 20240322stable)
Hash: da98001cfe5cc50cb2e9e4c241626d07dd4b06f5
Python 3.11.6, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-08-26:
Does this exist in other CAD programs?

luzpaz, 2024-08-26:
@pierreporte in realthunder's link branch: https://github.com/realthunder/FreeCAD/releases

pierreporte, 2024-08-26:
@luzpaz I mean other than FreeCAD and derivatives.

eblanshey, 2024-09-02:
@pierreporte I am only familiar with Solid Edge, and it does not have this option. Which I think I think is a positive for FreeCAD -- to not only be on par with other software, but to be better. This tolerance feature made 3d printed designs way easier to make.

pierreporte, 2024-09-02:
@eblanshey Thanks for the explanation. You point a very real problem with commonly used slicers: they don’t allow to alter geometry or define how different features of a part should be approached by the extruder. CAM programs can, and it makes perfect sense because it is not the role of the designer to prepare the geometry for manufacturing (this is different than designing with the manufacturing processes in mind).

A feature like this will not be enough to make FreeCAD better, and there are a lot of features missing to even pretend to be on par with commercial software. See the following board, especially P1 priority level: https://github.com/orgs/FreeCAD/projects/28.

FreeCAD is not getting an industry-grade slicer anytime soon. I think it could be interesting that the Mesh workbench allows to make offsets on surfaces so that the generated STL is ready for slicing while the source geometry remain unchanged.

eblanshey, 2024-09-02:
@pierreporte you make a good point about the slicers that should be able to make these adjustments, and I totally understand that there are more critical features needed in FreeCAD at the moment. I think having a dedicated place for this feature request still makes sense. Perhaps someone will want it badly enough to implement it themselves eventually.

slaesh, 2025-08-01:
> [@pierreporte](https://github.com/pierreporte) you make a good point about the slicers that should be able to make these adjustments, and I totally understand that there are more critical features [...]

I loved that feature too.. and I mean.. it is already implemented? xD dunno how far the realthunder fork is gone now.. but an experienced freecad dev could for sure "easily" copy/pick it :)

and yeah, its not crucial, but a few inputs here more wouldnt make such a big (negative) deal I guess :)

would love to see this coming :)

alanscodelog, 2026-04-28:
I was just trying to transition from realthunder's branch to v1.1 but ran into the lack of this feature which makes it impossible. I use this extensively. While in an ideal world as the designer you wouldn't have to do this, you end up doing it sometimes for the perfect fit.

Also even when slicers support adjusting fit, it's far from ideal because you might have problems with one part but not another. For example, with filament printers while this is usually something slicers support adjusting separately, circles need a larger tolerance than other parts. Another issue: a printed part against another printed part sometimes needs more tolerance than a printed part and a non-printed part. It depends a lot on the use case. I've had loads of parts I've had to fiddle with the fit of only one feature.

Even when not 3D printing this made it super easy to add tolerance in general because you can re-use a sketch for both the pad and the pocket if needed. And it has the plus that you can control how much and in what "direction" (does the pad get smaller or the hole wider), etc. Having something only for when generating meshes while interesting for other reasons, would be next to useless for my use cases and would not replace this feature for me.

Anyways, just my thoughts. I will try to investigate where this was added in the fork to see how feasible it would be to port it.