FreeCAD/FreeCAD #13129: FreeCAD starts, but cannot open existing drawings or create a new one

linuxdudeg, 2024-03-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I moved to a new Tuxedo Linux system and now FreeCAD does not work. It starts and the main menu's are active, but I cannot open an existing drawing or create a new one.
I get the MESA-LOADER: failed to open radeonsi error and have tried with the command line LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6 ./FreeCAD-0.21.2-Linux-x86_64.AppImage - does not help
I see this in the terminal window:
QOpenGLWidget: Failed to create context
QOpenGLWidget: Failed to create context
QOpenGLWidget: Failed to create context
No valid GL context found!
I tried using software OpenGL - does not help
I also have this error:
cannot access /home/xxxxx/.local/share/FreeCAD/Mod/ExtremeProlight
When I start and click on New I get

[code]
(qt.qpa.backingstore) composeAndFlush: QOpenGLContext creation failed
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
No valid GL context found!
[/code]

I am running X11, not Wayland

### Full version info

```shell
[code]
OS: TUXEDO OS 2 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * package.xml
  * README.md
  * ExtremeProLight 1.6.0
  * LICENSE
  * INSTALL.md
  * resources
  * screenshot.png
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

solonovamax, 2024-04-25:
I am experiencing the same issue.

I am currently using the `freecad-git` AUR package, so the latest commit, built from source. (ie. not the appimage)

the window does not update, but is still interactable. (ie. it will not redraw anything. resizing the window shows that it is frozen.)

*however*, right clicking causes a context menu to still appear. also, pressing <kbd>Ctrl + O</kbd> allows me to open a file. after opening the file and attempting to exit, it asks if I want to save the file.

so, it seems that it is still responsive & processing events, such as opening files. however, the window is *not* being redrawn.

I can give system specs if needed.

adrianinsaval, 2024-04-30:
are you running in a waylanda session? does the rpoblem persist with the latest weekly?

solonovamax, 2024-05-03:
> are you running in a waylanda session? does the rpoblem persist with the latest weekly?

I was on X11 while experiencing the issue. however, I believe it was caused due to not rebooting after nvidia driver updates (nvidia moment), as a reboot fixed it.
I can't speak for OP, though.

adrianinsaval, 2024-05-03:
I'll close since I can't reproduce either, OP please reopen if you are still experiencing this

closed 2024-05-03 by adrianinsaval

wagenadl, 2025-11-07:
I am experiencing the sample version with Freecad 1.0.2 on Ubuntu 24.04.3 LTS running X11 on NVidia RTX 3060 and NVidia driver 535.247.01.