FreeCAD/FreeCAD #13131: [Problem] V0.22 Sketcher_CreateFillet: clicking in an empty area removes the cursor tool icon

Roy-043, 2024-03-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Start Sketcher_CreateFillet.
2. Click in an empty area of the sketch.
3. Result: the cursor changes to a default arrow. But the tool is still active.
4. Move the cursor over an axis or some existing geometry and the cursor changes to a white cross with the tool icon again.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36487 (Git)
Build type: Release
Branch: main
Hash: f5baef888c25937026ee50001a4fe4293a9f11ee
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-25:
This is the case for many tools in Sketcher on my side. Happens also with dimension tool.

FEA-eng, 2024-07-01:
@PaddleStroke

PaddleStroke, 2024-07-03:
This bug is annoying indeed. I noticed it before with various tools.

PaddleStroke, 2024-07-03:
First observations : 
- This happen with all the sketcher DSH, including line, rectangle etc.. Clicking removes the cursor.
- Hovering on top of a line is re-applying the cursor by doing in several places : 
```
                if (edit->sketchHandler)
                    edit->sketchHandler->applyCursor();
```
Looking at the git blame these snippets are 13 years old so nothing new here.
- This was not happening in 0.20

closed 2024-07-15 by chennes