FreeCAD/FreeCAD #13132: Sketcher: dimensions can be interacted with by the label only

pierreporte, 2024-03-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Interaction with a dimension in a sketch is only possible with its label:

* it is preselected only when the label is hovered,
* it is selected only when the label is clicked,
* it is edited only when the label is double-clicked.

It would be a great improvement if the whole dimensions could be used to interact with it: dimension line (the arrow) and extension lines. It would be a solution to #12756.

### Full version info

```shell
OS: Ubuntu 23.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36526 (Git) AppImage
Build type: Release
Branch: main
Hash: 2ebbd83268ffabf41ceaad802714a4ffd9214437
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * BillOfMaterials-WB 0.0.7
  * CosmeticThread3D
  * TabBar (Disabled)
  * Curves 0.6.29
  * CurvedShapes 1.0.8
  * sheetmetal 0.4.6
  * OpenDark 2023.12.17
  * OpenLight 2023.12.17
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-03-24:
@FlachyJoe We briefly discussed this but you weren’t convinced. I tried to explain better what I wanted.

pierreporte, 2024-04-22:
@maxwxyz for the DWG project.

MisterMakerNL, 2025-04-08:
I think the best way is the label lets you move everything.
The line where the label is on, lets you move the dimenson up and down.
The side lines do nothing. Maybe select the dimension. Maybe even double clicking anywhere lets you edit the dimension.

[image]

pierreporte, 2025-04-08:
@MisterMakerNL I agree. Moving the label does the same thing as today, and moving the dimension line does a new thing. I can’t see what moving extension lines should do. In any case, clicking or double-clicking on any part of the dimension should select or trigger edit dialog respectively.

closed 2025-08-03 by kadet1090