FreeCAD/FreeCAD #13139: Limited GUI support for adding properties while designing

pieterhijma, 2024-03-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The workflow in FreeCAD is suboptimal for parametric design.  While designing, the user may want to create a parameter, for example for a Pad on a Sketch.  As soon as the user decides how to name the parameter, the user has to temporarily abort the Pad task and go to some document object or a spreadsheet to create the parameter.

In case of a document object, the user has to choose a document object, create a property which of itself has a limited workflow as discussed in #13038.  In case of a spreadsheet, the user has to go to the spreadsheet of choice, create a cell with a name for the parameter, create another cell (preferably close by, for example the next column) with the value, then add an alias to the cell that matches the name of the first cell (making sure the names are the same).

Then for both of these situations, the user has to remember how to access these variables (the path to the variables), going into the Pad task again, click on the expression symbol and type the path of the expression (for example `file#Spreadsheet.Height`).

The main issue is that users cannot add a parameter while still in the design task, therefore discouraging parametric design.

An obvious solution to this workflow problem is to create the parameters before doing the design.  However, unless it is very clear beforehand what the constraints of the design are, it is common to encounter situations that would benefit from having parameters.  These situations are subsequently hampered by the limitations of workflow just described.

### Full version info

```shell
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36330 +9 (Git)
Build type: DEBUG
Branch: direct-properties
Hash: a1801e61b5cf79b8eaacb3093d2f771c2cf78190
Python 3.11.6, Qt 5.15.12, Coin 4.0.2, Vtk , OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * OpenDark 2023.12.17
  * Ondsel-Lens 2024.1.22.01
  * backup 0.2.3
  * BIM 2021.12.0
  * Assembly3 0.12.2
  * Manipulator 1.5.7
  * fasteners 0.5.10
  * Assembly4 0.50.7
  * DynamicData 2.60.0
```

### Subproject(s) affected?

Core

### Anything else?

#### Context

- This issue is related to #12120 that gives an overview on custom data elements issues.
- This problem is related to #13038 discussing limited GUI support for custom data elements.
- This problem is discussed in [this Ondsel blogpost](https://ondsel.com/blog/build-system-for-custom-data-elements), specifically in [this section](https://ondsel.com/blog/build-system-for-custom-data-elements/#what-the-solution-should-look-like) that lists the following requirement:
  - Adding new data elements from Sketcher and other WBs should be simple and obvious.

#### Related

The Sketcher workbench has limited support to add properties while designing: It allows the user to name a constraint.  However, this is not a general approach that works beyond the Sketcher workbench, the entered name is not a direct property of the Sketch, but a property inside the property Constraints within the Sketch.  Moreover, the user cannot choose where to store the parameter, it has to reside in the Constraints property within the Sketch.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2024-03-25:
FYI I've started a currently named _Expression Manager_ macro to list all the expressions of a document, another tab will show named constraints.
It could has a third tab to list custom properties but it seems there is currently **no way to know if a property is user created**.

ref https://forum.freecad.org/viewtopic.php?t=86378

pieterhijma, 2024-03-25:
Thanks for letting me know. I think this comment is more fitting to issue #13038. Looks great but it seems we think alike. I've just created PR #13112. Please let me know what you think.

PR #13626 (2024-06-03): Gui: Add VarSet support to expression dialog
This allows users to directly add properties to variable sets from within the expression input dialog improving a workflow for parameterized design.

Closes #13139

A video that shows the functionality:

https://github.com/FreeCAD/FreeCAD/assets/5336482/5e0b6b8e-97a4-498f-bdc9-ce649089c55a
closed 2024-06-03 by chennes