FreeCAD/FreeCAD #13141: Some FEM mesh icons should be improved

FEA-eng, 2024-03-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Following #13125, there are 3 FEM mesh icons that should IMO be improved:

1. [Mesh group](https://wiki.freecad.org/FEM_MeshGroup)

[image]

This tool creates groups of mesh entities based on geometric entities (nodes on this face belong to this group, on that face to that group and so on). The icon shows this but I think that it could be shown on a single mesh object (big cube) - as a similar red outline including some elements.

2. [Mesh region](https://wiki.freecad.org/FEM_MeshRegion)

[image]

This tool is used for mesh refinement. The icon doesn't show it. The red color should cover elements along the whole edge and possibly include changing mesh density (more elements in that red region). Something like the next icon.

3. [Mesh boundary layer](https://wiki.freecad.org/FEM_MeshBoundaryLayer)

[image]

This one is not bad but too similar to what Mesh region should represent. IMO the icon should show that it's meant for flows. I would choose a simple 2D case like a square pipe:

[image]

@maxwxyz FYI

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36391 (Git)
Build type: Release
Branch: main
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-25:
@FEA-eng feedback for the first 2: 
[image]

FEA-eng, 2024-03-25:
> @FEA-eng feedback

Looks really great, thank you. Just one small thing - the FEM_MeshRegion icon should have the refinement and red highlight only on the two main rows surrounding that vertical edge:

[image]

maxwxyz, 2024-03-25:
@FEA-eng better?
[image]

FEA-eng, 2024-03-25:
> @FEA-eng better? 

Sure, perfect 👍

closed 2024-04-01 by sliptonic