FreeCAD/FreeCAD #13144: 0.22dev crashes on file creation on Wayland

katepavlu, 2024-03-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD crashes when creating a file. Happens on Ubuntu 23.10 on Wayland but not on X11.

Crash message:
`FreeCAD 0.22.0, Libs: 0.22.0devR36534 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

QSocketNotifier: Can only be used with threads started with QThread
QOpenGLFunctions created with non-current context
(qt.qpa.wayland) Wayland does not support QWindow::requestActivate()
freecad-daily: ./src/glue/gl.cpp:2354: const cc_glglue* cc_glglue_instance(int): Assertion current_ctx && "Must have a current GL context when instantiating cc_glglue!! (Note: if you are using an old Mesa GL version, set the environment variable COIN_GL_NO_CURRENT_CONTEXT_CHECK to get around what may be a Mesa bug.)"' failed.`

### Full version info

```shell
OS: Ubuntu 23.10 (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36534 (Git)
Build type: Release
Branch: main
Hash: d5b90e50af6a758af748179f289bb8f09e357266
Python 3.11.6, Qt 5.15.10, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Core

### Anything else?

QT_QPA_PLATFORM is not set to wayland ( [forum post suggesting this](https://forum.freecad.org/viewtopic.php?t=75251) )
Setting COIN_GL_NO_CURRENT_CONTEXT_CHECK does not fix the issue
Setting software OpenGL in  Preferences does not fix the issue.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

katepavlu, 2024-03-25:
Weekly build appimage does not crash

```
OS: Ubuntu 23.10 (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36526 (Git) AppImage
Build type: Release
Branch: main
Hash: 2ebbd83268ffabf41ceaad802714a4ffd9214437
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

katepavlu, 2024-03-25:
Opening existing files in daily build also causes the crash

katepavlu, 2024-03-26:
Interesting: crash also happens when I compile 2ebbd83268 (same commit as the working weekly appimage) from scratch

I think it's probably a dependency issue

katepavlu, 2024-03-26:
Backtrace:
```
[Thread debugging using libthread_db enabled]                                                                                                                                                 
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `./FreeCAD'.
Program terminated with signal SIGABRT, Aborted.
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:44
Download failed: Invalid argument.  Continuing without source file ./nptl/./nptl/pthread_kill.c.                                                                                              
44	./nptl/pthread_kill.c: No such file or directory.
[Current thread is 1 (Thread 0x7c3adec0da80 (LWP 52126))]
(gdb) 
(gdb) 
(gdb) 
(gdb) 
(gdb) 
(gdb) 
(gdb) 
(gdb) 
(gdb) bt
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=<optimized out>, signo=signo@entry=6) at ./nptl/pthread_kill.c:89
#3  0x00007c3adea428e6 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#4  0x00007c3adea268b7 in __GI_abort () at ./stdlib/abort.c:79
#5  0x00007c3adea267db in __assert_fail_base
    (fmt=0x7c3adebc5168 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", assertion=assertion@entry=0x7c3ade688518 "current_ctx && \"Must have a current GL context when instantiating cc_glglue!! (Note: if you are using an old Mesa GL version, set the environment variable COIN_GL_NO_CURRENT_CONTEXT_CHECK to get aroun"..., file=file@entry=0x7c3ade6d7811 "./src/glue/gl.cpp", line=line@entry=2354, function=function@entry=0x7c3ade6884e8 "const cc_glglue* cc_glglue_instance(int)") at ./assert/assert.c:92
#6  0x00007c3adea39206 in __assert_fail
    (assertion=0x7c3ade688518 "current_ctx && \"Must have a current GL context when instantiating cc_glglue!! (Note: if you are using an old Mesa GL version, set the environment variable COIN_GL_NO_CURRENT_CONTEXT_CHECK to get aroun"..., file=0x7c3ade6d7811 "./src/glue/gl.cpp", line=2354, function=0x7c3ade6884e8 "const cc_glglue* cc_glglue_instance(int)")
    at ./assert/assert.c:101
#7  0x00007c3ade3f6b18 in cc_glglue_instance (contextid=1) at /usr/src/coin3-4.0.0+ds-5/src/glue/gl.cpp:2354
#8  0x00007c3ade274798 in SoGLRenderActionP::isDirectRendering (this=this@entry=0x60d25d65ce00, state=state@entry=0x60d2603892e0)
    at /usr/src/coin3-4.0.0+ds-5/src/actions/SoGLRenderAction.cpp:1673
#9  0x00007c3ade27677a in SoGLRenderActionP::render (this=0x60d25d65ce00, node=0x60d2603d7a50) at /usr/src/coin3-4.0.0+ds-5/src/actions/SoGLRenderAction.cpp:1719
#10 0x00007c3ade26e40e in SoAction::apply (this=0x60d260661050, root=0x60d2603d7a50) at /usr/src/coin3-4.0.0+ds-5/src/actions/SoAction.cpp:565
#11 0x00007c3ae41add8c in Gui::SoBoxSelectionRenderAction::apply(SoNode*) () at /home/kat/source/repos/freecad-source/build/lib/libFreeCADGui.so
#12 0x00007c3ae42e21a8 in Gui::View3DInventorViewer::renderScene() () at /home/kat/source/repos/freecad-source/build/lib/libFreeCADGui.so
#13 0x00007c3ae42e19e5 in Gui::View3DInventorViewer::actualRedraw() () at /home/kat/source/repos/freecad-source/build/lib/libFreeCADGui.so
#14 0x00007c3ae42802cf in SIM::Coin3D::Quarter::QuarterWidget::paintEvent(QPaintEvent*) () at /home/kat/source/repos/freecad-source/build/lib/libFreeCADGui.so
#15 0x00007c3ae428a964 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::paintEvent(QPaintEvent*) () at /home/kat/source/repos/freecad-source/build/lib/libFreeCADGui.so
#16 0x00007c3ae01aeb28 in QWidget::event (this=this@entry=0x60d26063b550, event=event@entry=0x7ffe3af10c30) at kernel/qwidget.cpp:9045
#17 0x00007c3ae025ae52 in QFrame::event (this=0x60d26063b550, e=0x7ffe3af10c30) at widgets/qframe.cpp:550
#18 0x00007c3ae4280533 in SIM::Coin3D::Quarter::QuarterWidget::viewportEvent(QEvent*) () at /home/kat/source/repos/freecad-source/build/lib/libFreeCADGui.so
#19 0x00007c3adf4cc513 in QCoreApplicationPrivate::sendThroughObjectEventFilters (receiver=receiver@entry=0x60d26048b970, event=event@entry=0x7ffe3af10c30)
    at kernel/qcoreapplication.cpp:1190
#20 0x00007c3ae016bc72 in QApplicationPrivate::notify_helper (this=<optimized out>, receiver=0x60d26048b970, e=0x7ffe3af10c30) at kernel/qapplication.cpp:3634
#21 0x00007c3ae3d49b3c in Gui::GUIApplication::notify(QObject*, QEvent*) () at /home/kat/source/repos/freecad-source/build/lib/libFreeCADGui.so
#22 0x00007c3adf4cc7a8 in QCoreApplication::notifyInternal2 (receiver=0x60d26048b970, event=0x7ffe3af10c30) at kernel/qcoreapplication.cpp:1064
#23 0x00007c3adf4cc9a2 in QCoreApplication::sendSpontaneousEvent (receiver=<optimized out>, event=<optimized out>) at kernel/qcoreapplication.cpp:1474
#24 0x00007c3ae01a6c69 in QWidgetPrivate::sendPaintEvent (this=this@entry=0x60d2606aca20, toBePainted=...) at kernel/qwidget.cpp:5479
#25 0x00007c3ae01cf2e0 in QOpenGLWidget::event (this=0x60d26048b970, e=0x7ffe3af11010) at kernel/qopenglwidget.cpp:1473
#26 0x00007c3ae428221d in CustomGLWidget::event(QEvent*) () at /home/kat/source/repos/freecad-source/build/lib/libFreeCADGui.so
#27 0x00007c3ae016bc82 in QApplicationPrivate::notify_helper (this=<optimized out>, receiver=0x60d26048b970, e=0x7ffe3af11010) at kernel/qapplication.cpp:3640
#28 0x00007c3ae3d49b3c in Gui::GUIApplication::notify(QObject*, QEvent*) () at /home/kat/source/repos/freecad-source/build/lib/libFreeCADGui.so
[... cut ...]
#33 0x00007c3ae01ab579 in QWidgetPrivate::showChildren (this=this@entry=0x60d2603966b0, spontaneous=spontaneous@entry=false) at kernel/qwidget.cpp:8209
#34 0x00007c3ae01ab613 in QWidgetPrivate::show_helper (this=0x60d2603966b0) at kernel/qwidget.cpp:7794
#35 0x00007c3ae01ab40f in QWidgetPrivate::show_recursive (this=<optimized out>) at kernel/qwidget.cpp:7718
#36 0x00007c3ae01ab599 in QWidgetPrivate::showChildren (this=this@entry=0x60d26053b6a0, spontaneous=spontaneous@entry=false)
    at ../../include/QtWidgets/../../src/widgets/kernel/qwidget.h:132
```

katepavlu, 2024-03-26:
Running the binary from the weekly appimage with my system dependencies instead of the bundled ones causes it to start crashing too

katepavlu, 2024-04-08:
@maxwxyz what else is needed for diagnosis?

xtemp09, 2024-04-09:
Looks like it's connected to #9273, coin3d/coin#451 and void-linux/void-packages#14626.

blacklightpy, 2024-05-13:
> Looks like it's connected to https://github.com/FreeCAD/FreeCAD/issues/9273, https://github.com/coin3d/coin/issues/451 and https://github.com/void-linux/void-packages/issues/14626.

And https://github.com/void-linux/void-packages/pull/37027

ii8, 2024-05-20:
There is an existing issue for this: #6288
In fact there are two but this one has been closed:  #6068 (there was no fix, it was just a misunderstanding and shouldn't have been closed)

maxwxyz, 2024-10-07:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/6288

closed 2024-10-07 by maxwxyz