FreeCAD/FreeCAD #13146: Draft command line-like input

FEA-eng, 2024-03-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It certainly has been requested on the forum in the past but I couldn't find any GitHub issue for this so let me create one and present my idea to resolve this. 2D CAD software typically allows for quick drawing using command line input. AutoCAD is known for that but even LibreCAD has it:

[image]

I know that implementing full command line input for Draft would require a lot of work so I'd like to suggest way less laborious approach - just an option to merge the input fields for points into a single input field where one can enter all 3 coordinates (possibly with an option to skip the third one so that it's just 0 or current Z coordinate) at once using some separator (to be discussed):

[image]

This would be especially useful when adding multiple individual points (one could even paste the coordinates from some file) but also when drawing rectangles and other shapes in Draft.

@Roy-043 FYI

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36391 (Git)
Build type: Release
Branch: main
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-03-25:
There has been a lot of discussion about this and also some tries, that have ended with https://github.com/FreeCAD/FreeCAD/issues/7078

I have some doubts about that "unified" field a lot because it requires a much larger task panel column, and renders a lot of current Draft system complicated to reimplement, such as greying out some of the coordinates when you restrict movement. Also, using different units becomes problematic.

All in one, that seems to me a lot of work for very limited benefit... You are aware that you can press Enter to jump between the different coordinate fields right? For me that feels a lot like command-line, you can complete a Draft command completely from the keyboard, without touching the mouse

FEA-eng, 2024-03-25:
> All in one, that seems to me a lot of work for very limited benefit... You are aware that you can press Enter to jump between the different coordinate fields right? For me that feels a lot like command-line, you can complete a Draft command completely from the keyboard, without touching the mouse

Yes, I know. I thought that it would be relatively easy to implement something like this (or at least much easier than the whole command line). Entering coordinates this way is IMO very convenient when there are multiple geometric entities to draw (and my workflows often involve checking some exact point coordinates and pasting them to create points so this would be way easier as well). Other software even has it in sketchers (there it could be useful too but Paddle's on-view parameters are fine for now).

[image]

I agree that there would be some issues to consider (like units - I assume lack thereof) but maybe someone will share my view and implement this in the future, hence the GitHub issue.