FreeCAD/FreeCAD #13150: Error when deleting first digit of circle radius in Task view

furgo16, 2024-03-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Load the Draft workbench
2. Click on the circle button on the toolbar to start drawing a circle 
3. Click on a point on the view to define the circle's center point
4. Use the <kbd>Del</kbd> keyboard key to delete the value on the Radius input box on the Task window.
5. When deleting from last to first digit, upon deleting the first one the following error is shown on the Report view:
 
```python traceback
20:17:50  Traceback (most recent call last):
20:17:50    File "/snap/freecad/966/usr/Mod/Draft/DraftGui.py", line 1100, in checkSpecialChars
20:17:50      if txt[0] in "0123456789.,-":
20:17:50  IndexError: string index out of range
```

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36534 (Git) Snap 966
Build type: Release
Branch: main
Hash: d5b90e50af6a758af748179f289bb8f09e357266
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * BIM 2021.12.0
  * Assembly4 0.50.12
  * ExplodedAssembly
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-03-25:
@yorikvanhavre the traceback seems to point to a line of code introduced in [this commit](https://github.com/FreeCAD/FreeCAD/commit/b2f1add784bda38eef0a4c98cf31e183d5de3411)

maxwxyz, 2024-03-25:
can confirm

luzpaz, 2024-03-25:
For reference, the code in main:  
https://github.com/FreeCAD/FreeCAD/blob/380d452adb5da85f47543a316b6b533f84e829c7/src/Mod/Draft/DraftGui.py#L1098-L1102

luzpaz, 2024-04-19:
@Roy-043 would you mind weighing-in ?

PR #13543 (2024-04-19): Draft: Fix error when deleting first digit of input
Fixes #13150.
commit 64bcdbf9 (2024-04-19): Draft: Fix error when deleting first digit of input

Fixes #13150.
closed 2024-04-19 by Roy-043