FreeCAD/FreeCAD #13151: Regression Obj Outputs (Alias and WaveFront) in 0.21.2 -> 0.21.1 OK

boucaron, 2024-03-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi,

At the export, there is a problem in the produced Obj files.  
There is an issue with the normals.

I tried in both 0.21.2 and  0.21.2. 
I do not have the issue with the 0.21.1 at the Export.

Steps to reproduce: 
- Open the provided file
- Export the Cut in Obj (either the Alias one or the Wavefront one) 
- Then Import as Obj (Alias or Wavefront)

From 0.21.2
[image]
[image]

[image]

Ok in 0.21.1
[image]

The example is attached to the issue

Regards,
Julien

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * Curves 0.6.29
[/code]
```

### Subproject(s) affected?

None

### Anything else?

[RegressionObjOutput.zip](https://github.com/FreeCAD/FreeCAD/files/14749610/RegressionObjOutput.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-26:
can you test if the error persists in the current dev builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

boucaron, 2024-03-26:
Tested with current dev build
The issue is still present export from Wavefront Obj and reimport in either Alias or Wavefront OBJ
One case a bit better for Wavefront Obj, but still incomplete (highlighted in green)

[image]

FEA-eng, 2025-06-22:
Still relevant

FEA-eng, 2025-10-25:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/21 12:07:13
Build type: Release
Branch: main
Hash: 2096d0bd8603b429fdc0c746c34b3d73c3289fe1
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

luzpaz, 2025-11-25:
@boucaron can you test on 1.2-dev (AKA weekly) ? https://github.com/FreeCAD/FreeCAD/releases

Roy-043, 2025-11-29:
Problem also occurs in this recent weekly:
```
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.2.0dev.20251118 (Git shallow)
Build date: 2025/11/18 21:54:51
Build type: Release
Branch: main
Hash: be7d41dea75f7f2966d7b431d24f7060c6907339
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.4, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Turntable/Window center
Stylesheet/Theme/QtStyle: Tree_Lines.qss/FreeCAD Classic/
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```