FreeCAD/FreeCAD #13152: Crash on Redo

RiccaDS, 2024-03-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD crashes when hitting Redo, both with Crlt+Shift+Z or from the menu Edit->Redo. 
This happens very often, almost always, to the point that I do not use it anymore. I think this happens however after some operations have been done, because I cannot reproduce this voluntarily with a freshly opened file. It is therefore hard to give a reproducible action to generate the problem.

### Full version info

```shell
This also happens on 0.20 and 0.21 appimages

OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36487 (Git) AppImage
Build type: Release
Branch: main
Hash: f5baef888c25937026ee50001a4fe4293a9f11ee
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * sheetmetal 0.4.4
  * Render 2024.1.23
  * Assembly4 0.50.12
  * fasteners 0.5.14
  * workfeature-macro
  * CurvedShapes 1.0.8
  * FeedsAndSpeeds 0.5.0
  * freecad.gears 1.0.0
  * Curves 0.6.29
  * OSE3dPrinter 0.1.0
  * Design456 0.0.1
  * kicadStepUpMod 11.1.0
  * QuickMeasure 2022.10.28
  * pcb 6.2023.1
  * Launcher
  * ExplodedAssembly
  * toSketch 1.0.1
  * A2plus 0.4.64
  * Manipulator 1.5.7
  * MakerWorkbench 1.0.1
  * Estimate 0.1.2
  * workfeature
  * FEMbyGEN 2.1.0
```

### Subproject(s) affected?

None

### Anything else?

Very hard to tell the context in which this happens. If I am given instructions on how to generate a log file I can try troubleshooting. 
Issue has been discussed here https://forum.freecad.org/viewtopic.php?style=10&t=73643

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

RiccaDS, 2024-03-26:
Issue is experienced by me and was experienced by the OP in the forum post I linked.

luzpaz, 2024-03-26:
Please add the example file to the ticket (zip it and drag drop it into github text window. TIA

xtemp09, 2024-04-07:
@RiccaDS, what are the reproduction steps? Have you tried to run _FreeCAD_ without addons?

RiccaDS, 2024-04-19:
> @RiccaDS, what are the reproduction steps? Have you tried to run _FreeCAD_ without addons?

Reproducibility is very hard to log in this issue as it makes this problem very often but not always and when it crashes, unless you casually saved the file just before, you will lose the user actions performed up until that moment.  I have now been able to save a file before it crashes but not on REDO, it crashes on UNDO(Ctrl+Z) this time.
Use the attached file and
1- disable the thread in the threaded rod like in the pic. 
2- left click on the viewport
3-Ctrl+Z
If it does not crash the first time, try again the same steps. 
[image]

[crash2.zip](https://github.com/FreeCAD/FreeCAD/files/15046043/crash2.zip)

xtemp09, 2024-04-20:
No crash here.

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36882 (Git)
Build type: Debug
Branch: main
Hash: cf1eeaab19da2d2a6eaaba3ae6bd1fc8ad620552
Python 3.10.12, Qt 5.15.12, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

----

@RiccaDS, when a program crashes, _Linux_ outputs backtrace into the command-line. Please, upload it here.

Also, launch _FreeCAD_ with logging on [^1] and upload the log file here.

[^1]: https://wiki.freecad.org/Debugging

RiccaDS, 2024-04-20:
@xtemp09 as you can see from the video I had to do the operation a few times before it crashed.
```
free(): invalid next size (normal)
/tmp/.mount_FreeCAq5TKdS/AppRun: line 43: 436310 Aborted                 (core dumped) "${MAIN}" "$@"
```
[FreeCAD.log](https://github.com/FreeCAD/FreeCAD/files/15048724/FreeCAD.log)

https://github.com/FreeCAD/FreeCAD/assets/101465894/8e7511d6-1ce8-420a-8d0d-a2f32105d7b4

luzpaz, 2024-04-20:
There are 2 issues mentioned that are 3rd party addon related:

1. `Assembly4 0.50.12` not being compatible with your version of FreeCAD v0.22
2. `OSE3dPrinter 0.1.0` throws issues

luzpaz, 2024-11-15:
@RiccaDS is this issue still present for you in RC4 ?

RiccaDS, 2024-11-17:
> @RiccaDS is this issue still present for you in RC4 ?

I am working on 1.1dev 39092 right now. I can confirm issue persists. Also, this is another laptop I set up in the last few weeks with another version of Ubuntu and with fresh FC appimages and preferences.
I experienced at least 3 crashes in the last few days, on different projects, and following Ctrl+Shift+Z. Last one I had it right now when working on A2Plus WB. But this seems WB unrelated. 
```
OS: Ubuntu 24.10 (ubuntu:GNOME/ubuntu/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39092 (Git) AppImage
Build type: Release
Branch: main
Hash: a65d05c6e495038168eefcba6a1e43218fbb3a20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Curves 0.6.51
  * Estimate 0.1.2
  * fasteners 0.5.29
  * A2plus 0.4.68

```

RiccaDS, 2024-11-17:
ok trying to use Ctrl+Shift+Z more often to test it out, maybe the crash doesn't happen all that often, however I can say the problem is still there.

maxwxyz, 2025-02-12:
@ryanthara if you have steps to reliably reproduce the crash on macOS please add them here.

ryanthara, 2025-02-12:
@maxwxyz hard to detect from my perspective.

I have a file with a small VarSet, a couple of Bodys and not so much complicated things.

During writing this FC crashed again.

My attempt:

- new file (without saving)
- new Body
- new Sketch with a hexagon
- Pad it to 10 mm
- create a SubShapeBinder of the Pad
- Pad this SupShapeBinder
- set Offset to -10 (never done before)
- and hit control + Z

Boom! Crashes on an Intel MBP with current macOS.

After restarting it, the restore dialog occurs.

Here the output of the console.

`21:53:46  flag
21:53:46  flag2
21:53:46  (qt.qpa.fonts) Populating font family aliases took 471 ms. Replace uses of missing font family "Sans" with one that exists to avoid this cost. 
21:56:44  Reading failed from embedded file: Body.Shape.bin (0 bytes, 2 bytes compressed)
21:56:44  Reading failed from embedded file: Sketch.InternalShape.bin (0 bytes, 2 bytes compressed)
21:56:44  <ElementMap> ElementMap.cpp(476): No hasherRef
21:56:44  Reading failed from embedded file: Sketch.InternalShape.bin (0 bytes, 2 bytes compressed)
21:56:44  <ElementMap> ElementMap.cpp(476): No hasherRef
21:56:44  <ElementMap> ElementMap.cpp(476): No hasherRef
21:56:44  Reading failed from embedded file: Pad.SuppressedShape.bin (0 bytes, 2 bytes compressed)
21:56:44  <ElementMap> ElementMap.cpp(476): No hasherRef
21:56:44  Reading failed from embedded file: Sketch001.InternalShape.bin (0 bytes, 2 bytes compressed)
21:56:44  Reading failed from embedded file: Revolution.SuppressedShape.bin (0 bytes, 2 bytes compressed)
21:56:44  Reading failed from embedded file: Sketch002.InternalShape.bin (0 bytes, 2 bytes compressed)
21:56:44  <ElementMap> ElementMap.cpp(476): No hasherRef
21:56:44  <ElementMap> ElementMap.cpp(476): No hasherRef
21:56:44  Reading failed from embedded file: Pocket.SuppressedShape.bin (0 bytes, 2 bytes compressed)
21:56:44  <ElementMap> ElementMap.cpp(476): No hasherRef
21:56:44  Reading failed from embedded file: PolarPattern.SuppressedShape.bin (0 bytes, 2 bytes compressed)
21:56:44  Reading failed from embedded file: Sketch003.InternalShape.bin (0 bytes, 2 bytes compressed)
21:56:44  <ElementMap> ElementMap.cpp(476): No hasherRef
21:56:44  <ElementMap> ElementMap.cpp(476): No hasherRef
21:56:44  Reading failed from embedded file: Pocket001.SuppressedShape.bin (0 bytes, 2 bytes compressed)
21:56:44  Reading failed from embedded file: Sketch004.InternalShape.bin (0 bytes, 2 bytes compressed)
21:56:44  <ElementMap> ElementMap.cpp(476): No hasherRef
21:56:44  <ElementMap> ElementMap.cpp(476): No hasherRef
21:56:44  Reading failed from embedded file: Pocket002.SuppressedShape.bin (0 bytes, 2 bytes compressed)
21:56:44  <ElementMap> ElementMap.cpp(476): No hasherRef
21:56:44  Reading failed from embedded file: MultiTransform.SuppressedShape.bin (0 bytes, 2 bytes compressed)
21:56:44  Reading failed from embedded file: PolarPattern001.Shape.bin (0 bytes, 2 bytes compressed)
21:56:44  Reading failed from embedded file: PolarPattern001.SuppressedShape.bin (0 bytes, 2 bytes compressed)
21:56:44  Reading failed from embedded file: Sketch006.InternalShape.bin (0 bytes, 2 bytes compressed)
21:56:44  <ElementMap> ElementMap.cpp(476): No hasherRef
21:56:44  <ElementMap> ElementMap.cpp(476): No hasherRef
21:56:44  Reading failed from embedded file: Pocket003.SuppressedShape.bin (0 bytes, 2 bytes compressed)
21:56:44  <ElementMap> ElementMap.cpp(476): No hasherRef
21:56:44  Reading failed from embedded file: MultiTransform001.SuppressedShape.bin (0 bytes, 2 bytes compressed)
21:56:44  Reading failed from embedded file: PolarPattern002.Shape.bin (0 bytes, 2 bytes compressed)
21:56:44  Reading failed from embedded file: PolarPattern002.SuppressedShape.bin (0 bytes, 2 bytes compressed)
21:56:44  Reading failed from embedded file: PolarPattern003.Shape.bin (0 bytes, 2 bytes compressed)
21:56:44  Reading failed from embedded file: PolarPattern003.SuppressedShape.bin (0 bytes, 2 bytes compressed)
21:56:44  Reading failed from embedded file: Sketch007.InternalShape.bin (0 bytes, 2 bytes compressed)
21:56:44  <ElementMap> ElementMap.cpp(476): No hasherRef
21:56:44  <ElementMap> ElementMap.cpp(476): No hasherRef
21:56:44  Reading failed from embedded file: Pocket004.SuppressedShape.bin (0 bytes, 2 bytes compressed)
21:56:44  <ElementMap> ElementMap.cpp(476): No hasherRef
21:56:44  Reading failed from embedded file: MultiTransform002.SuppressedShape.bin (0 bytes, 2 bytes compressed)
[... 74 lines cut ...]
21:56:46  Reading failed from embedded file: Clone.SuppressedShape.bin (0 bytes, 2 bytes compressed)
21:56:46  <ElementMap> ElementMap.cpp(476): No hasherRef
21:56:46  <ElementMap> ElementMap.cpp(476): No hasherRef
21:56:46  Reading failed from embedded file: Clone001.SuppressedShape.bin (0 bytes, 2 bytes compressed)
21:56:46  <ElementMap> ElementMap.cpp(476): No hasherRef
21:56:46  <ElementMap> ElementMap.cpp(476): No hasherRef
21:56:46  Reading failed from embedded file: Clone002.SuppressedShape.bin (0 bytes, 2 bytes compressed)
21:56:46  <ElementMap> ElementMap.cpp(476): No hasherRef
21:56:46  Reading failed from embedded file: Sketch019.InternalShape.bin (0 bytes, 2 bytes compressed)
21:56:46  <ElementMap> ElementMap.cpp(476): No hasherRef
21:56:46  <ElementMap> ElementMap.cpp(476): No hasherRef
21:56:46  Reading failed from embedded file: Pad009.SuppressedShape.bin (0 bytes, 2 bytes compressed)
21:56:46  Reading failed from embedded file: Sketch020.InternalShape.bin (0 bytes, 2 bytes compressed)
[... cut ...]
21:56:46  Reading failed from embedded file: Revolution.SuppressedShape.bin (0 bytes, 2 bytes compressed)
21:56:46  Reading failed from embedded file: Sketch001.InternalShape.bin (0 bytes, 2 bytes compressed)
21:56:46  Reading failed from embedded file: Pad.SuppressedShape.bin (0 bytes, 2 bytes compressed)
21:56:46  Reading failed from embedded file: Sketch002.InternalShape.bin (0 bytes, 2 bytes compressed)
21:56:46  Reading failed from embedded file: Pad001.SuppressedShape.bin (0 bytes, 2 bytes compressed)
`

luzpaz, 2025-11-29:
@johnocfii would you mind testing https://github.com/FreeCAD/FreeCAD/issues/13152#issuecomment-2654829262 on macOS ?

luzpaz, 2025-11-29:
Thanks for the very thorough reponse!

@ryanthara and @RiccaDS please retest on latest dev build

Lunarien, 2026-10-22:
I am on the 2nd release candidate of 26.3 and i still occasionally get a pop up saying "Access violation", although there is no crash. It only really seems to happen after a while of using FreeCAD, though im not entirely sure. It happens on every redo after it happens for the first time (though the redo still works despite the pop up. i just close the pop up and continue.). Restarting FreeCAD always gets rid of the popups for a while.