FreeCAD/FreeCAD #13161: TechDraw: 'Template' object brings nothing to be stand-alone

PaddleStroke, 2024-03-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]
In techdraw when you click on insert default page, you get a page and a child template object.

I do not see why page and template and made separate objects. It brings no functionality and just clutter the tree.
Suggestion : 
[image]

Forum discussion : https://forum.freecad.org/viewtopic.php?t=86271

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-26:
Is it possible to copy the template object to another page or FC document?

PaddleStroke, 2024-03-26:
But that wouldn't make much sense to do so. As the template is basically a link to the SVG file. You can just set the link to the same.

WandererFan, 2024-03-26:
Maybe it is my background, but I would not spend scarce devel resources on this.  It works, doesn't cause any real problems, why change it?

PaddleStroke, 2024-03-26:
I don't say it's a priority, but a small unecessary complexity.

Of course it is not a big thing and doesn't cause problem. But it's useless and clutters hence increasing software complexity, even if a little bit. And small streams make big rivers.

So if it's not hard to fix, why not doing it? If it's very big or annoying then indeed it may not be worth the hassle.

maxwxyz, 2024-04-21:
Maybe related if you use an editor to edit the page:
- https://github.com/FreeCAD/FreeCAD/issues/13577

closed 2024-10-30 by WandererFan