FreeCAD/FreeCAD #13166: Sketcher Symmetry tool should be renamed to Mirror for clarity

FEA-eng, 2024-03-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Symmetry tool in the Sketcher has always been named like this but it can be confusing since we have the Symmetry constraint too. And the first thing one looks for to mirror sketch geometry is in fact Mirror. Don't you think that renaming this tool would make it more clear? Just the button and menu option, it could stay as it is internally. It shouldn't be problematic in terms of backward compatibility or something because it's a tool that is applied once, does its thing and leaves no special objects in the tree or other traces.

@PaddleStroke Do you agree with this or not necessarily?

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36391 (Git)
Build type: Release
Branch: main
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-03-26:
There is already a mirror sketch tool that works outside of edit mode. So having two tools with the same name would be confusing too.

Besides the symmetry tool does not only mirror because it can also do symmetry via point.

I'm not much confused with the symmetric constraint.
That's just my opinion though, curious to see what others think about it.

FEA-eng, 2024-03-26:
> There is already a mirror sketch tool that works outside of edit mode. So having two tools with the same name would be confusing too.

Right, but it's not within the sketch mode tools so I treat it separately. But it might be necessary to consider this as well. Maybe Mirror and Mirror sketch are fine though as it's clear that they are used for different purposes.

> Besides the symmetry tool does not only mirror because it can also do symmetry via point.

Yeah, the new version can mirror about a point too (if it can be called this way but seems to be ok). I'm thinking about this as a mirror tool applying symmetry constraints. My point is mainly that the operation is commonly known as mirroring in other CAD software so new users may not find it easily.

> That's just my opinion though, curious to see what others think about it.

It would be interesting to ask some new users (maybe on Discord or here). I guess that veterans are used to this and may not want it like many other changes of existing tools.

FEA-eng, 2024-03-26:
Interestingly, CATIA has both Mirror and Symmetry tools in the sketcher. The difference is only that the former keeps the original geometry while the latter removes it. Something we can do with a single tool in FreeCAD. Other CAD software that I know has just the Mirror tool and keeps both instances or also offers an option to keep or remove the original geometry (like the Mirror Entities tool in SolidWorks).