FreeCAD/FreeCAD #13167: FEM fails to compile under MSVC

chennes, 2024-03-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When compiling from current main (26 March 2024), FEM fails to compile under MSVC, complaining:
```
src\Mod\Fem\App\FemPostObject.cpp(138): error C2872: 'vtkWriter': ambiguous symbol
LibPack-0.21-V2.10\include\vtk-8.2\vtkWriter.h(42): note: could be 'vtkWriter'
src\Mod\Fem\App\FemPostObject.cpp(78): note: or       'void `anonymous-namespace'::vtkWriter(const char *,const vtkSmartPointer<vtkDataObject> &)'
src\Mod\Fem\App\FemPostObject.cpp(138): error C2059: syntax error: '<'
```
The anonymous namespace can be resolved using `::vtkWriter`, but that still leaves me with a strange syntax error (that doesn't look like an error to me).

### Full version info

```shell
n/a can't compile
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-03-26:
@marioalexis84 Can it be caused by #12987 ?

marioalexis84, 2024-03-26:
> Can it be caused by https://github.com/FreeCAD/FreeCAD/pull/12987

It shouldn't be the problem. The CI system compiles without problems using the same VTK version as the chennes system.

@chennes did you try reinstalling VTK on your system?
I've had strange compiler problems before. Reinstalling the conflicting library resolved the issue. I assume the failure was due to some corrupted file.

chennes, 2024-03-26:
I do not have VTK installed, this is using the Windows LibPack. I did try downloading a new version of the LibPack, but the problem persists.

marioalexis84, 2024-03-26:
perhaps a bug in that specific version of MSVC? You can change the name of the function template vtkWriter. When I wrote it I didn't realize that it coincided with the name of a VTK object. However, using the same name can be confusing.

chennes, 2024-03-27:
OK, how about `femVTKWriter` instead?

marioalexis84, 2024-03-27:
perfect

PR #13174 (2024-03-27): FEM: Rename vtkWriter to femVTKWriter to make MSVC happy
Fixes #13167
closed 2024-03-27 by wwmayer