FreeCAD/FreeCAD #13169: Can't deactivate or delete constraints in Sketcher WB.

mfraser, 2024-03-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Using FreeCAD 0.22~pre1~202403261451~ubuntu22.04.1, I am unable to delete an existing constraint in the sketcher WB. If I hi-light the constraint I want to delete, nothing happens and as soon as I move off the constraint it is unselected. I can't even delete them from the constraint list. If I try to deactivate a constraint from the list, I'm told "Wrong Selection. Select constraints from sketch."

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36617 (Git)
Build type: Release
Branch: main
Hash: 4845d391ed2702b0b51a9da905066485eb5ef7bc
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * QuickMeasure 2022.10.28
  * kicadStepUpMod 11.1.0
  * CurvedShapes.backup1710693524.3928182 1.0.5 (Disabled)
  * fasteners 0.5.15
  * Curves 0.6.29
  * ThreadProfile 1.89.0
  * sheetmetal 0.4.6
  * VendorParts
  * lattice2 1.0.0
  * Freecad-Built-in-themes-beta 1.2.2
  * Lithophane
  * ProDarkThemePreferencePack 1.0.0
  * Manipulator 1.5.7
  * DynamicData 2.60.0
  * Defeaturing 1.2.2
  * CurvedShapes 1.0.8
  * Assembly4 0.12.4
  * freecad.gears 1.2.0
  * A2plus 0.4.64
  * fcgear 1.0.0
```

### Subproject(s) affected?

Sketcher

### Anything else?

FreeCAD_Weekly AppImage version below works. So somewhere between 36617 and 36526, something stopped working.

```
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36526 (Git) AppImage
Build type: Release
Branch: main
Hash: 2ebbd83268ffabf41ceaad802714a4ffd9214437
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * QuickMeasure 2022.10.28
  * kicadStepUpMod 11.1.0
  * CurvedShapes.backup1710693524.3928182 1.0.5 (Disabled)
  * fasteners 0.5.15
  * Curves 0.6.29
  * ThreadProfile 1.89.0
  * sheetmetal 0.4.6
  * VendorParts
  * lattice2 1.0.0
  * Freecad-Built-in-themes-beta 1.2.2
  * Lithophane
  * ProDarkThemePreferencePack 1.0.0
  * Manipulator 1.5.7
  * DynamicData 2.60.0
  * Defeaturing 1.2.2
  * CurvedShapes 1.0.8
  * Assembly4 0.12.4
  * freecad.gears 1.2.0
  * A2plus 0.4.64
  * fcgear 1.0.0
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-27:
I cannot reproduce, how are you trying to delete them, in the 3D view or in the task dialog list? Via `del`-key or with the context menu in the 3D view?
Could you try it in the latest build: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
There is no other tool active?

mamu7211, 2024-03-27:
Can reproduce it, I tried to create a "Slot" using Part Design, New (Document) > New Body (in Part Design) > New Sketch (on XZ)  > Slot Tool > click first center point on X Axis, the other one also on the X Axis.

Both center points are constrained with "Constraint point onto object" at the X axis, when trying to delete one of the "Constraint point onto object" following things are observed:

- click on the constraint does not register, constraint is highlighted when hovering but not selected
- click on the constraint in the Constraints toolbox does select the constraint but `del`-key does not delete it
- right click on the constraint in the Constraints toolbox and selecting "Delete" does not work as intended, the constraint is not deleted
- might be related or be a new issue, but ... it is also not possible to add new constraints to the geometry without creating solver errors.

Maybe related?? Axis selection (X/Y) and center point selection does not work. When I create a Slot with center points not attached to an axis and adding a radius constraint after creation, the Radius constraint is there but cannot be deleted.

[image]

Creating a Slot with both center points attached to an axis and applying a constraint after creation of the Slot, a "Redundant constraints: (8)" solver message is shown.

[image]

...maybe this is connected or is a different issue I cannot tell, If you like I can create another issue.

## Errors

** Report View **
```
12:39:04  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Datum 10,62 mm for the constraint with index 8 is invalid
12:39:04  Unnamed#Sketch: App.getDocument('Unnamed').getObject('Sketch').setDatum(8,App.Units.Quantity('10.620000 mm'))
```

**Python Console**
```
Python 3.10.12 (main, Nov 20 2023, 15:14:05) [GCC 11.4.0] on linux
Type 'help', 'copyright', 'credits' or 'license' for more information.
>>> # Gui.runCommand('Sketcher_CompCreateCircle',0)
>>> # Gui.runCommand('Sketcher_CompSlot',0)
>>> lastGeoId = len(ActiveSketch.Geometry)
>>> 
>>> geoList = []
>>> geoList.append(Part.ArcOfCircle(Part.Circle(App.Vector(-25.966066, 0.000000, 0.000000), App.Vector(0.000000, 0.000000, 1.000000), 10.615082), 1.570796, 4.712389))
>>> geoList.append(Part.ArcOfCircle(Part.Circle(App.Vector(29.211678, 0.000000, 0.000000), App.Vector(0.000000, 0.000000, 1.000000), 10.615082), 4.712389, 7.853982))
>>> geoList.append(Part.LineSegment(App.Vector(-25.966066, 10.615082, 0.000000),App.Vector(29.211678, 10.615082, 0.000000)))
>>> geoList.append(Part.LineSegment(App.Vector(-25.966066, -10.615082, 0.000000),App.Vector(29.211678, -10.615082, 0.000000)))
>>> App.getDocument('Unnamed').getObject('Sketch').addGeometry(geoList,False)
>>> del geoList
>>> 
>>> constraintList = []
>>> constraintList.append(Sketcher.Constraint('Tangent', 0, 1, 2, 1))
>>> constraintList.append(Sketcher.Constraint('Tangent', 0, 2, 3, 1))
>>> constraintList.append(Sketcher.Constraint('Tangent', 1, 2, 2, 2))
>>> constraintList.append(Sketcher.Constraint('Tangent', 1, 1, 3, 2))
>>> constraintList.append(Sketcher.Constraint('Equal', 0, 1))
>>> constraintList.append(Sketcher.Constraint('Horizontal', 3))
>>> App.getDocument('Unnamed').getObject('Sketch').addConstraint(constraintList)
>>> del constraintList
>>> 
>>> constraintList = []
>>> constraintList.append(Sketcher.Constraint('PointOnObject', 0, 3, -1))
>>> constraintList.append(Sketcher.Constraint('PointOnObject', 1, 3, -1))
>>> App.getDocument('Unnamed').getObject('Sketch').addConstraint(constraintList)
>>> del constraintList
>>> 
>>> # Gui.Selection.addSelection('Unnamed','Body','Sketch.Edge1',-34.5538,-0.00799926,-6.23939,False)
>>> ### Begin command Sketcher_Dimension
>>> App.getDocument('Unnamed').getObject('Sketch').addConstraint(Sketcher.Constraint('Radius',0,10.615082)) 
>>> ### End command Sketcher_Dimension
>>> # Gui.Selection.clearSelection()
>>> App.getDocument('Unnamed').getObject('Sketch').setDatum(8,App.Units.Quantity('10.620000 mm'))
>>> 
```

## Version + Mods

OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36631 (Git) Snap 968
Build type: Release
Branch: main
Hash: dc3e0c3dcf56042530c64ca4e6c90da44c094597
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.60f
  * Assembly4 0.12.4
  * freecad.gears 1.0.0
  * fasteners 0.4.25

mfraser, 2024-03-27:
I updated to 202403271451 just now and that also has the issue, but I've gone back to 202403251447 and it is working again.

```
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36535 (Git)
Build type: Release
Branch: main
Hash: e5d4f0e38eb3a0c867d5ef383db081285ffeeba1
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * QuickMeasure 2022.10.28
  * kicadStepUpMod 11.1.0
  * fasteners 0.5.15
  * Curves 0.6.29
  * ThreadProfile 1.89.0
  * sheetmetal 0.4.7
  * VendorParts
  * lattice2 1.0.0
  * Freecad-Built-in-themes-beta 1.2.2
  * Lithophane
  * ProDarkThemePreferencePack 1.0.0
  * Manipulator 1.5.7
  * DynamicData 2.60.0
  * Defeaturing 1.2.2
  * CurvedShapes 1.0.8
  * Assembly4 0.12.4
  * freecad.gears 1.2.0
  * A2plus 0.4.64
  * fcgear 1.0.0
```
So, something after [commit e5d4f0e] (https://github.com/FreeCAD/FreeCAD/commit/e5d4f0e38eb3a0c867d5ef383db081285ffeeba1) caused this issue.

PaddleStroke, 2024-03-27:
I can replicate the issue.

Looking at commits coming after (https://github.com/FreeCAD/FreeCAD/commit/e5d4f0e38eb3a0c867d5ef383db081285ffeeba1), the ones that seems the more likely to be linked to this are the one from TNP work @bgbsww, like f19df465ea152f2bb3ec9068e72f7282a1b74b7c perhaps?

FlachyJoe, 2024-03-27:
A git bisect confirms the error is due to #13084 

@mamu7211 the redundant constraints error with slot is due to auto vertical/horizontal. You can open another issue.

bgbsww, 2024-03-28:
This is definitely #13084; working to track it down now.  Specifically the overridden getElementName is not doing what it should.  The rest of the PR seems to be okay, but we'll see.

bgbsww, 2024-03-28:
I think this is fixed with #13184.

closed 2024-03-28 by maxwxyz

mfraser, 2024-05-08:
This issue is back again.

```
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37213 (Git)
Build type: Release
Branch: main
Hash: 20e7deb86a8c6c2cd2378f09f8313760933f3a5c
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * QuickMeasure 2022.10.28
  * fasteners.backup1712686361.5869133 0.5.16 (Disabled)
  * kicadStepUpMod 11.1.4
  * fasteners 0.5.20
  * lattice2.backup1713204440.8139822 1.0.0 (Disabled)
  * Curves 0.6.36
  * ThreadProfile 1.89.0
  * sheetmetal 0.4.13
  * VendorParts
  * lattice2 1.0.0
  * kicadStepUpMod.backup1714893889.389688 11.1.3 (Disabled)
  * Freecad-Built-in-themes-beta 1.2.2
  * Lithophane
  * ProDarkThemePreferencePack 1.0.0
  * Curves.backup1714929565.0217447 0.6.35 (Disabled)
  * CurvedShapes.backup1713382123.4246979 1.0.8 (Disabled)
  * Manipulator 1.5.7
  * Defeaturing.backup1713382129.3425534 1.2.2 (Disabled)
  * DynamicData 2.62.0
  * Defeaturing 1.2.2
  * CurvedShapes 1.0.9
  * Assembly4 0.12.4
  * freecad.gears 1.2.0
  * A2plus 0.4.64a
  * fcgear 1.0.0
```

maxwxyz, 2024-05-16:
@bgbsww there was a mention on the forums that it's again not working
https://forum.freecad.org/viewtopic.php?p=758437#p758437

bgbsww, 2024-05-16:
Yeah, 'cause I broke it again, and then it was fixed again in #13910 this morning.