FreeCAD/FreeCAD #13171: Treeview disappears while in techdraw page view

decryptedchaos, 2024-03-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I explicitly prefer the TreeView over the combo view,  but using it with TechDraw is very painful to the point that you have to either switch back to the part design page to add new views to the page or enable combo view..  

Here is a video of the behavior 

https://github.com/FreeCAD/FreeCAD/assets/1302391/4419aeeb-9cb6-4ae9-b1a2-0b117288f96d

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.64
  * Assembly4 0.50.9
  * Dracula 0.0.5
  * Glass
  * Marz 0.0.31-alpha
  * Render 2024.1.23
  * sheetmetal 0.4.3
[/code]
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-28:
could you please try it in the weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

luzpaz, 2024-03-31:
@decryptedchaos ping

WandererFan, 2024-04-01:
Seems to work here.  Something missing in the problem description?
[image]

decryptedchaos, 2024-04-03:
I will check this shortly, sorry i don't check github daily..   I usually check for replies 24 hours after my post, and then my ADHD kicks in and i forget..  

> could you please try it in the weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

I suspect it works here because you have it floating,  mine is docked and i suspect the window gets crated differently programmatically  docked vs floating.  

> Seems to work here. Something missing in the problem description? ![TreeViewWithTDPage](https://private-user-images.githubusercontent.com/3587161/318462949-7a29e6d6-eb15-495f-a0ed-e70acbc89aa4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTIxMzc3NTcsIm5iZiI6MTcxMjEzNzQ1NywicGF0aCI6Ii8zNTg3MTYxLzMxODQ2Mjk0OS03YTI5ZTZkNi1lYjE1LTQ5NWYtYTBlZC1lNzBhY2JjODlhYTQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQwMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MDNUMDk0NDE3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Nzc3OGY4ZDg1MWVkYjgyNTg0Yjk3ZTM2YjYzZjViMThhNTkyYzRlMmEzM2U5NzA5NGZkMzkzYjRhYmU5Y2VjNSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Tzi7MjT_QxKRuo6GyzNzgKdh8FzxD0b-mWjt3S3vZ9E)

decryptedchaos, 2024-04-03:
Yes, I can confirm that it does the same thing on the weekly snapshot.  It looks identical to the above posted video clip so I see no need to post another. 

Please note what I referenced above,  that my Tree view is docked as a panel (or whatever you call it when its transparently embedded on the left side like shown..  

For further insight I can show this,  when I'm in the normal Part Design view  Tree View panel is checked,  but when I go to the TechDraw page,  Tree View gets forcefully unchecked and **_cannot_** be checked as shown here..  then when returning to Part Design it suddenly rechecks itself.. 
 

[image]

luzpaz, 2024-07-18:
@decryptedchaos can you add the revision number of the weekly snapshot you've tested this on ?

luzpaz, 2024-08-24:
@decryptedchaos :point_up_2:

decryptedchaos, 2024-08-26:
I'll check again tomorrow,  i have no clue what the original snapshot i tested was it was on a different install/distro

decryptedchaos, 2024-09-06:
Just to give you a update on this, since i use Linux now,  i'm not entirely sure my analysis is still relevant,  but when i was going back and trying to setup a test environment to reproduce this,  i found that the issue might in fact be with the Glass add-on itself and not a core issue

luzpaz, 2024-09-06:
>  i found that the issue might in fact be with the Glass add-on itself and not a core issue

This would be another usecase for https://github.com/FreeCAD/FreeCAD/issues/15710

Ok, closing this ticket for now. If you can reproduce with latest dev version of RC1.0 please re-open.

closed 2024-09-06 by luzpaz