FreeCAD/FreeCAD #13173: Build fails with LTO

eli-schwartz, 2024-03-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I tried to build with the following *FLAGS to optimize the build: `-flto=4 -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing`

Note the -Werror=* flags are used to help detect cases where the compiler tries to optimize by assuming UB cannot exist in the source code -- if it does exist, ordinarily the code would be miscompiled, and this says to make the miscompilation a fatal error.

I got this error:
```
[2/1534] : && /usr/bin/x86_64-pc-linux-gnu-g++ -fPIC -Wall -Wextra -Wno-write-strings -march=native -fstack-protector-all -O2 -pipe -fdiagnostics-color=always -frecord-gcc-switches -flto=4 -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing  -Wformat -Werror=format-security -Wno-sign-compare -Wno-reorder -Wno-switch -Wno-unused-variable -Wno-unused-but-set-variable -Wno-comment -Wno-unused-parameter -Wno-empty-body -Wno-pedantic -Wno-unused-result -Wno-cast-function-type -Wno-maybe-uninitialized -Wno-missing-field-initializers  -Wl,-O1 -Wl,--as-needed -flto=4 -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing -Wl,--defsym=__gentoo_check_ldflags__=0 -shared -Wl,-soname,libMEFISTO2.so -o lib/libMEFISTO2.so src/3rdParty/salomesmesh/CMakeFiles/MEFISTO2.dir/src/MEFISTO2/aptrte.cpp.o src/3rdParty/salomesmesh/CMakeFiles/MEFISTO2.dir/src/MEFISTO2/trte.c.o -L/usr/lib64/opencascade -Wl,-rpath,/usr/lib64/opencascade:  -lmpi  /usr/lib64/opencascade/libTKOffset.so.7.7.0  /usr/lib64/opencascade/libTKFeat.so.7.7.0  /usr/lib64/opencascade/libTKSTL.so.7.7.0  /usr/lib64/opencascade/libTKBin.so.7.7.0  /usr/lib64/opencascade/libTKBinL.so.7.7.0  /usr/lib64/opencascade/libTKMeshVS.so.7.7.0  /usr/lib64/opencascade/libTKRWMesh.so.7.7.0  /usr/lib64/opencascade/libTKXDESTEP.so.7.7.0  /usr/lib64/opencascade/libTKXDEIGES.so.7.7.0  /usr/lib64/opencascade/libTKFillet.so.7.7.0  /usr/lib64/opencascade/libTKSTEP.so.7.7.0  /usr/lib64/opencascade/libTKSTEPAttr.so.7.7.0  /usr/lib64/opencascade/libTKSTEP209.so.7.7.0  /usr/lib64/opencascade/libTKSTEPBase.so.7.7.0  /usr/lib64/opencascade/libTKIGES.so.7.7.0  /usr/lib64/opencascade/libTKXSBase.so.7.7.0  /usr/lib64/opencascade/libTKBool.so.7.7.0  /usr/lib64/opencascade/libTKXCAF.so.7.7.0  /usr/lib64/opencascade/libTKVCAF.so.7.7.0  /usr/lib64/opencascade/libTKCAF.so.7.7.0  /usr/lib64/opencascade/libTKBO.so.7.7.0  /usr/lib64/opencascade/libTKPrim.so.7.7.0  /usr/lib64/opencascade/libTKV3d.so.7.7.0  /usr/lib64/opencascade/libTKMesh.so.7.7.0  /usr/lib64/opencascade/libTKShHealing.so.7.7.0  /usr/lib64/opencascade/libTKHLR.so.7.7.0  /usr/lib64/opencascade/libTKTopAlgo.so.7.7.0  /usr/lib64/opencascade/libTKGeomAlgo.so.7.7.0  /usr/lib64/opencascade/libTKBRep.so.7.7.0  /usr/lib64/opencascade/libTKGeomBase.so.7.7.0  /usr/lib64/opencascade/libTKG3d.so.7.7.0  /usr/lib64/opencascade/libTKG2d.so.7.7.0  /usr/lib64/opencascade/libTKService.so.7.7.0  /usr/lib64/opencascade/libTKMath.so.7.7.0  -lfontconfig  -lGL  -lX11  /usr/lib64/libfreetype.so  /usr/lib64/opencascade/libTKLCAF.so.7.7.0  /usr/lib64/opencascade/libTKCDF.so.7.7.0  /usr/lib64/opencascade/libTKXDE.so.7.7.0  /usr/lib64/opencascade/libTKernel.so.7.7.0  -lpthread  -lrt  -lstdc++  -ldl && :
FAILED: lib/libMEFISTO2.so 
: && /usr/bin/x86_64-pc-linux-gnu-g++ -fPIC -Wall -Wextra -Wno-write-strings -march=native -fstack-protector-all -O2 -pipe -fdiagnostics-color=always -frecord-gcc-switches -flto=4 -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing  -Wformat -Werror=format-security -Wno-sign-compare -Wno-reorder -Wno-switch -Wno-unused-variable -Wno-unused-but-set-variable -Wno-comment -Wno-unused-parameter -Wno-empty-body -Wno-pedantic -Wno-unused-result -Wno-cast-function-type -Wno-maybe-uninitialized -Wno-missing-field-initializers  -Wl,-O1 -Wl,--as-needed -flto=4 -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing -Wl,--defsym=__gentoo_check_ldflags__=0 -shared -Wl,-soname,libMEFISTO2.so -o lib/libMEFISTO2.so src/3rdParty/salomesmesh/CMakeFiles/MEFISTO2.dir/src/MEFISTO2/aptrte.cpp.o src/3rdParty/salomesmesh/CMakeFiles/MEFISTO2.dir/src/MEFISTO2/trte.c.o -L/usr/lib64/opencascade -Wl,-rpath,/usr/lib64/opencascade:  -lmpi  /usr/lib64/opencascade/libTKOffset.so.7.7.0  /usr/lib64/opencascade/libTKFeat.so.7.7.0  /usr/lib64/opencascade/libTKSTL.so.7.7.0  /usr/lib64/opencascade/libTKBin.so.7.7.0  /usr/lib64/opencascade/libTKBinL.so.7.7.0  /usr/lib64/opencascade/libTKMeshVS.so.7.7.0  /usr/lib64/opencascade/libTKRWMesh.so.7.7.0  /usr/lib64/opencascade/libTKXDESTEP.so.7.7.0  /usr/lib64/opencascade/libTKXDEIGES.so.7.7.0  /usr/lib64/opencascade/libTKFillet.so.7.7.0  /usr/lib64/opencascade/libTKSTEP.so.7.7.0  /usr/lib64/opencascade/libTKSTEPAttr.so.7.7.0  /usr/lib64/opencascade/libTKSTEP209.so.7.7.0  /usr/lib64/opencascade/libTKSTEPBase.so.7.7.0  /usr/lib64/opencascade/libTKIGES.so.7.7.0  /usr/lib64/opencascade/libTKXSBase.so.7.7.0  /usr/lib64/opencascade/libTKBool.so.7.7.0  /usr/lib64/opencascade/libTKXCAF.so.7.7.0  /usr/lib64/opencascade/libTKVCAF.so.7.7.0  /usr/lib64/opencascade/libTKCAF.so.7.7.0  /usr/lib64/opencascade/libTKBO.so.7.7.0  /usr/lib64/opencascade/libTKPrim.so.7.7.0  /usr/lib64/opencascade/libTKV3d.so.7.7.0  /usr/lib64/opencascade/libTKMesh.so.7.7.0  /usr/lib64/opencascade/libTKShHealing.so.7.7.0  /usr/lib64/opencascade/libTKHLR.so.7.7.0  /usr/lib64/opencascade/libTKTopAlgo.so.7.7.0  /usr/lib64/opencascade/libTKGeomAlgo.so.7.7.0  /usr/lib64/opencascade/libTKBRep.so.7.7.0  /usr/lib64/opencascade/libTKGeomBase.so.7.7.0  /usr/lib64/opencascade/libTKG3d.so.7.7.0  /usr/lib64/opencascade/libTKG2d.so.7.7.0  /usr/lib64/opencascade/libTKService.so.7.7.0  /usr/lib64/opencascade/libTKMath.so.7.7.0  -lfontconfig  -lGL  -lX11  /usr/lib64/libfreetype.so  /usr/lib64/opencascade/libTKLCAF.so.7.7.0  /usr/lib64/opencascade/libTKCDF.so.7.7.0  /usr/lib64/opencascade/libTKXDE.so.7.7.0  /usr/lib64/opencascade/libTKernel.so.7.7.0  -lpthread  -lrt  -lstdc++  -ldl && :
/var/tmp/portage/media-gfx/freecad-9999/work/freecad-9999/src/3rdParty/salomesmesh/inc/aptrte.h:304:3: error: type of ‘teajte_’ does not match original declaration [-Werror=lto-type-mismatch]
  304 |   teajte( Z & mxsomm, Z &  nbsomm, R_3 * mnpxyd,  R_3 * comxmi,
      |   ^
/var/tmp/portage/media-gfx/freecad-9999/work/freecad-9999/src/3rdParty/salomesmesh/src/MEFISTO2/trte.c:1719:22: note: return value type mismatch
 1719 | /* Subroutine */ int teajte_(integer *mxsomm, integer *nbsomm, doublereal *
      |                      ^
/var/tmp/portage/media-gfx/freecad-9999/work/freecad-9999/src/3rdParty/salomesmesh/src/MEFISTO2/trte.c:1719:22: note: type ‘int’ should match type ‘void’
/var/tmp/portage/media-gfx/freecad-9999/work/freecad-9999/src/3rdParty/salomesmesh/src/MEFISTO2/trte.c:1719:22: note: ‘teajte_’ was previously declared here
/var/tmp/portage/media-gfx/freecad-9999/work/freecad-9999/src/3rdParty/salomesmesh/src/MEFISTO2/trte.c:1719:22: note: code may be misoptimized unless ‘-fno-strict-aliasing’ is used
/var/tmp/portage/media-gfx/freecad-9999/work/freecad-9999/src/3rdParty/salomesmesh/inc/aptrte.h:316:3: error: type of ‘tehote_’ does not match original declaration [-Werror=lto-type-mismatch]
  316 |   tehote( Z & nutysu, Z & nbarpi, Z &  mxsomm, Z &  nbsomm, R_3 * mnpxyd,
      |   ^
/var/tmp/portage/media-gfx/freecad-9999/work/freecad-9999/src/3rdParty/salomesmesh/src/MEFISTO2/trte.c:1989:22: note: return value type mismatch
 1989 | /* Subroutine */ int tehote_(integer *nutysu, integer *nbarpi, integer *
      |                      ^
/var/tmp/portage/media-gfx/freecad-9999/work/freecad-9999/src/3rdParty/salomesmesh/src/MEFISTO2/trte.c:1989:22: note: type ‘int’ should match type ‘void’
/var/tmp/portage/media-gfx/freecad-9999/work/freecad-9999/src/3rdParty/salomesmesh/src/MEFISTO2/trte.c:1989:22: note: ‘tehote_’ was previously declared here
/var/tmp/portage/media-gfx/freecad-9999/work/freecad-9999/src/3rdParty/salomesmesh/src/MEFISTO2/trte.c:1989:22: note: code may be misoptimized unless ‘-fno-strict-aliasing’ is used
/var/tmp/portage/media-gfx/freecad-9999/work/freecad-9999/src/3rdParty/salomesmesh/inc/aptrte.h:331:3: error: type of ‘tetrte_’ does not match original declaration [-Werror=lto-type-mismatch]
  331 |   tetrte( R_3 * comxmi, R & aretmx, Z & nbarpi, Z & mxsomm, R_3 * mnpxyd,
      |   ^
/var/tmp/portage/media-gfx/freecad-9999/work/freecad-9999/src/3rdParty/salomesmesh/src/MEFISTO2/trte.c:2460:22: note: return value type mismatch
[... cut ...]
 * Checking out https://github.com/FreeCAD/FreeCAD.git to /var/tmp/portage/media-gfx/freecad-9999/work/freecad-9999 ...
git checkout --quiet refs/git-r3/HEAD
GIT update -->
   repository:               https://github.com/FreeCAD/FreeCAD.git
   at the commit:            dc3e0c3dcf56042530c64ca4e6c90da44c094597
```
```

### Subproject(s) affected?

None

### Anything else?

Originally reported downstream: https://bugs.gentoo.org/875221
Full build log: [build.log](https://github.com/FreeCAD/FreeCAD/files/14765936/build.log)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-04-15:
Hi @adrianinsaval, is this on your radar ?

kstar, 2024-06-19:
I reproduce the same issue on my ArchLinux machine.

luzpaz, 2024-07-24:
@eli-schwartz @kstar is this still reproducible on latest git main branch ?

eli-schwartz, 2024-07-25:
@luzpaz yes, I just did a fresh build and the issue is indeed still there.

To be fair, I wouldn't expect it to go away without some fairly deliberate action. Additionally, the code with the issues was last modified on or before ee22ac24c495d1fea301fa1e6a824a6a7fb55933 which was over a year before the original date this ticket was filed.

luzpaz, 2024-07-27:
Heads up @petterreinholdtsen, would you have a moment to look in to this ?

petterreinholdtsen, 2024-07-27:
[luzpaz]
> Heads up @petterreinholdtsen, would you have a moment to look in to
> this ?

Unfortunately I do not currently have the required free disk space to
test.  Are you unsure how to bring the prototypes in sync?

-- 
Happy hacking
Petter Reinholdtsen

luzpaz, 2024-10-24:
cc @adrianinsaval

eli-schwartz, 2025-06-17:
@stalebot not stale

(I have no idea whether it is or not. I'm not doing daily builds so I can provide daily confirmation that it still applies. If your stalebot doesn't put any effort into figuring out whether the issue applies, I'm baffled that anyone thinks the mere presence of such a bot will make ME put in vast efforts to prove that it doesn't.)

https://nostalebots.xyz/
https://blog.benwinding.com/github-stale-bots/
https://fvsch.com/stale-bots

oursland, 2025-08-27:
@stalebot not stale

luzpaz, 2025-09-28:
> It could be that these were intended to be file-local static types, which such a declaration would resolve the issue.

Anyone want to submit a fix ?