FreeCAD/FreeCAD #13175: 0.21.2 (Windows) - Sketcher - Arc of ellipse by center - ERROR

PeMiHolm, 2024-03-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Windows 10/11: 
- Sketcher 
- Arc of ellipse

Can't complete, makes distorted ellipse 

### Full version info

```shell
When trying to use - in Sketcher - Arc of ellipse:
0. Click the function OK
1. 1st click - Placing start of arc OK
2. 2nd click - Placing vertical radius OK
3. Before 3rd click - trying to set horizontal radius - line starts jumping / flickering
4. 3rd click - Placing horizontal radius get all wrong
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-27:
I cannot reproduce, could you try this on the current dev version: 
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

PeMiHolm, 2024-03-27:
Hey Max...

If.. I've got some pics (printscreens) showing the error....

Thanks for the link...
FreeCAD_weekly-builds-36596-2024-03-25-conda-Windows-x86_64-py311.7z
<https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-36596-2024-03-25-conda-Windows-x86_64-py311.7z>
1. Zip to extract ... ?
2. Then just put the folders & files over the existing files ?

Is That the way to go ?

Regards

/Per/

maxwxyz, 2024-03-27:
extract zip and open bin folder, there is a freecad.exe.
Please also add your full about info in your issue.

PeMiHolm, 2024-03-27:
I added the full weekly update FreeCAD 0.22 
- unzip
- placed all files on top of existing 0.21.2
(just the \\bin\FreeCAD.exe gave windows 10 error, FreeCad did not start)

Same problem in both releases.
Work around: (see example below)
- Nr 2. instruction - see that you are away from vertical centerline (just by little)
- Then after last instruction adjust your vertical centerline
- Good to go...

When trying to use - in Sketcher - Arc of ellipse:
0. Click the function OK
1. 1st click - Placing start of arc OK
2. 2nd click - Placing vertical radius OK
3. Before 3rd click - trying to set horizontal radius - line starts jumping / flickering
4. 3rd click - Placing horizontal radius get all wrong

Roy-043, 2024-03-27:
Confirmed. The problem occurs if the 2nd point has the exact same X coordinate as the 1st. You have to snap to the grid to ensure that is the case.

Roy-043, 2024-10-08:
Snapping to the grid causes the problem, it is not a workaround.

luzpaz, 2024-10-14:
Anybody willing to work on this ?

FEA-eng, 2025-09-27:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

Lokestrom, 2026-02-01:
This issue is caused by a numerical singularity in the math used to compute the ellipse parameters when seeking the third point. Snapping is not the cause, it only makes the issue easier to trigger.

### Problem:
When the first and second points have the same X coordinate (vertical major axis), `atan2(dy, dx)` correctly returns `±π/2`. However, this angle is later used in expressions involving `tan(phi)` and divisions by `cos(phi)`.

For `phi = ±π/2`, `cos(phi) → 0` and `tan(phi)` becomes unbounded, leading to severe numerical instability. This causes the ellipse to behave erratically when moving the third point.

### Location
`DrawSketchHandlerArcOfEllipse::mouseMove(SnapManager::SnapHandle snapHandle)`
`DrawSketchHandlerArcOfEllipse.h`

I have confirmed that the issue is still present on the current `main` branch.

I believe this can be fixed by reformulating the computation using direction vectors (avoiding `tan(phi)` / division by `cos(phi)`), 
I would like to propose a patch.