FreeCAD/FreeCAD #13177: External geometry lines cannot be selected

komoricodrutz, 2024-03-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi everyone,

Adding an external geometry from another sketch inside the same body works. However, when trying to select a line, nothing happens. For example, trying to add an equality constraint to that external geometry line gives the message that at least two lines should be selected.
Deleting the external geometry lines also doesn't work. Nothing happens.
The vertexes, however, are usable, so adding point to line constraints, for example, when using a point from the external geometry, works just fine.
Issue wasn't present yesterday.

https://github.com/FreeCAD/FreeCAD/assets/134259571/9d31363d-4c53-4138-891e-da5ce318bacd

### Full version info

```shell
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36617 (Git)
Build type: Release
Branch: main
Hash: 4845d391ed2702b0b51a9da905066485eb5ef7bc
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * BOLTSFC 2022.11.5
  * A2plus 0.4.60n
  * Lithophane
  * Assembly4 0.50.2
  * Glass
  * fasteners 0.4.62
  * fcgear 1.0.0
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ifreislich, 2024-03-27:
In addition to the above, I cannot select the origin or axes when adding constraints in the Sketcher. For example constrain point onto object allows to select point but the cursor changes to forbidden when over the axes. I can start a geometry with point on object or coincident on the origin or axes.

OS: Ubuntu 23.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36633 (Git)
Build type: Unknown
Branch: main
Hash: 360e280c64c56f1a083645f566cc4125330703f8
Python 3.11.6, Qt 5.15.10, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
  * fcgear 1.0.0

ifreislich, 2024-03-27:
Introduced by 55acedb83d0a25f7d2721572b3018ae503aea75b

maxwxyz, 2024-03-27:
@realthunder @bgbsww FYI

kaiwas, 2024-03-27:
Could this problem be related to this one?
https://github.com/FreeCAD/FreeCAD/issues/13009

bgbsww, 2024-03-28:
Hmm.  I only get a still from the example mp4 above; if I try to open it in a separate window I get an error that the file is no longer available.

I'm trying to replicate this by:
1) Opening a new Document
2) Opening PartDesign
3) Creating a sketch in the XY Plane
4) Closing that sketch
5) Creating a new sketch in the XY Plane
6) Sketch -> Sketcher Geometries -> Create External Geometry
7) Selecting several of the edges from the first sketch.
8) External geometry is created.
9) Trying to select the external geometry fails.

This is using last week's build from https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds that predates #13084.   There are clearly issues with that PR, but I'm not sure this is one.  Happy to be wrong about that, but right now I'm confused - there appears to be another cause for this issue.

OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36526 (Git) AppImage
Build type: Release
Branch: main
Hash: 2ebbd83268ffabf41ceaad802714a4ffd9214437
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * FreeCAD_BearingsWB 0.0.2
  * Assembly4 0.50.4
  * fcgear 1.0.0
  * Plot 2022.4.17
  * DynamicData 2.54.0
  * BIM 2021.12.0
  * parts_library

ifreislich, 2024-03-28:
I'm building from source.  Replicate the unable to select origin or axes adding constraints:

git checkout f19df465ea152f2bb3ec9068e72f7282a1b74b7c
build/install
1. Start page -> Parametric part
2. Create sketch in XY plane
3. draw a line anywhere
4.1 Choose create point onto object constraint
4.2 Select any end of the line.
4.3 Select either axis <---- cannot perform this action.

git checkout 2712a98d47dd23dae1d0c5828305c4ca739efaf0 (previous ref)
build/install
Steps above work as expected.

To arrive at f19df46 I did a binary search recompiling until I found the commit which introduced the issue I'm observing. I'm just a user so I can't say how the commit changed the behavior.

commit 55acedb83d0a25f7d2721572b3018ae503aea75b
Author: bgbsww <bgbsww@gmail.com>
Date:   Sun Mar 24 18:13:21 2024 -0400

    Clean, enable disabled element map code in sketcher, add tests

commit f19df465ea152f2bb3ec9068e72f7282a1b74b7c **<----- Does not work after here**
Author: Zheng, Lei <realthunder.dev@gmail.com>
Date:   Thu Mar 21 15:17:32 2024 -0400

    Toponaming/Part: Move sketcher override of getElementName over

commit 2712a98d47dd23dae1d0c5828305c4ca739efaf0    **<----- Works up to here**
Author: wandererfan <wandererfan@gmail.com>
Date:   Sun Mar 24 11:44:12 2024 -0400

    [TD]fix ComplexSection with ViewDirection == Z

commit a171886f8cb701506dce1521b92d1960036a695d
Author: wandererfan <wandererfan@gmail.com>
Date:   Mon Mar 25 11:52:16 2024 -0400

    [TD]remove 2 unused methods

bgbsww, 2024-03-28:
Thanks!  The select origin or axis issue appears to be resolved with #13184.  However, the reported external geometry lines selection issue appears to still exist and predate #13084.

komoricodrutz, 2024-03-28:
Hi,
I just started playing with it again.
In addition to the initial symptoms, there are several more things which cannot be selected inside a sketch:
Centerpoint (for instance if you wish to set two vertexes symmetrical to the center point)
X and Y axes (same as above)
Apparently any other constraint cannot be selected in the sketch. Until now: Parallel, Point to Object, Perpendicular, Tangent, Horizontal, Vertical.
Constraints for Angle and Length can only be double-clicked for editing.
Deleting any constraint is out of the question.
Even when selecting a constraint from the constraints list, they cannot be deleted.

<details>
<summary>Daily version Build Details</summary>

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36633 (Git)
Build type: Release
Branch: main
Hash: 360e280c64c56f1a083645f566cc4125330703f8
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * BOLTSFC 2022.11.5
  * A2plus 0.4.64
  * Lithophane
  * Assembly4 0.50.12
  * Glass
  * fasteners 0.5.15
  * fcgear 1.0.0

```

</details>

komoricodrutz, 2024-03-30:
Hi. Seems all these issues have been fixed now in 36677.
Thank you!

<details>
<summary>Daily version Build Details</summary>

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36677 (Git)
Build type: Release
Branch: main
Hash: 3fadb1f40bdfa4ee1c61acbf12fbe44680723965
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * BOLTSFC 2022.11.5
  * A2plus 0.4.64
  * Lithophane
  * Assembly4 0.50.12
  * Glass
  * fasteners 0.5.15
  * fcgear 1.0.0

```

</details>

CalligaroV, 2024-03-30:
Hi Everyone!

I was trying too to reproduce this issue but had no luck, the logic keeps working correctly (I'm at commit 36680)

@komoricodrutz, if you too can perform the operation of this issue without errors I would say that either close it or wait if someone else has something to add.

Besides that, are you referring only to this issue or to #13009 too?

maxwxyz, 2024-03-30:
I cannot reproduce either, will close it.

closed 2024-03-30 by maxwxyz

samohd, 2024-08-23:
I have this issue as of 38495. I can add the external geometry lines... and they appear dashed purple. However, if I click on the line, nothing seems to change in the GUI to show it's selected. Using the Elements panel, external geometry lines can be selected... and deleted if you wish. But they are ghosted so they can't be unchecked as a way to tell which line is which if you have multiple external geometry lines. If you only have one, it's obvious which it is cuz it's the one that's ghosted.

maxwxyz, 2024-08-23:
Can you add a demo file / screenshots and more info what version and theme you are using?

samohd, 2024-08-23:
I am brand new to FreeCAD... I'm only using the default setup mostly with build 38495 on Windows 10. I started 2 or 3 builds ago in the past week.

I've dabbled a little with the GUI/Theme selection, but this is a bit of a mess, so I've reset the settings mostly. I currently have Prefs > General > General Theme = FreeCAD Light. Prefs > Display > UI > Style Sheet = FreeCAD Light -and- Overlay style sheet = Auto (Default). I just tried changing the Style Sheet = FreeCAD Dark... and also tried Theme = FreeCAD Classic, and neither of those make a difference.

I can confirm that if I click on an external geometry line, and press DELETE on keyboard, the line is deleted. The only visual acknowledgement that it's selected though is that the Model > Data panel populates.

Weird stuff happens like the Tasks panel's title bar disappears when its not under the mouse. Doesn't matter what Hide Mode is set to. Sometimes I've seen that the text in panels and the bg color are the same... unless I mouse over them, or select them. But, I reset or changed theme back and forth a couple times, and that's back to normal now I think. The Tasks panel bg can only have a little color at the top (if not set to transparent)... I think it fades to white at the bottom. Though I have Prefs > Display > Colors >Background Color set to Simple. I think this is just the drawing window area, not the panels or style sheets or whatever they're called. Notably, it seems to me that the Model and Tasks panes are using a different GUI engine, as things like making them wider are done in different ways.

I'm not sure what Overlay is referring to in Prefs > Display > UI settings. Seems like these are maybe Model, or Tasks, but maybe not both. They aren't really overlays, so maybe not.

Not sure what you mean by a demo file... presumably GUI settings aren't stored in .FCStd projects are they? I do notice that if I change the theme in Prefs, an opened Parts document indicates its been changed... which seems weird.

[image]

CalligaroV, 2024-09-07:
@samohd, I confirm your issue about the selection/highlighting of the external geometry.

I opened a PR that looks fixing it, at least on my tests. (PR #16355)
Can you try it and see if this fixes the issue for you too?

TIA!

samohd, 2024-09-07:
@CalligaroV  I'm not sure how to check it from that PR link. I'm about to install latest build 38643, might the fix be in that?

CalligaroV, 2024-09-07:
Can you compile FreeCAD?
If so you could checkout/download my branch with this modification and test it (https://github.com/CalligaroV/FreeCAD/archive/refs/heads/Sketcher-checkSubName-from-LS3.zip)

About your second question no, this fix will be available also in newer builds only after (and if) it will be merged.
A check made by someone else will help to decide if it makes sense to merge it, that's why I pinged you.

samohd, 2024-09-07:
I've not delved into dev on FC yet other than a little Python macroing. Still working my way thru all the bugs I've encountered doing my first project in FC. Formerly having used TinkerCAD only.

samohd, 2024-09-07:
Have you got a compiled module you want me to copy into my regular build and try? Windows x64.

CalligaroV, 2024-09-07:
I see.

Unfortunately all I can share are the modified files and/or my branch (AFAIK).
Moreover, I have only a Linux machine and no machines/VM with Windows, sorry.

Any case, this modification touches C++ files, therefor to see its effect it is necessary to compile FreeCAD.

If you can't compile then we may ask someone else to test it