FreeCAD/FreeCAD #13178: FCDtd file format backwards compatibility issue (files created in 0.22 seem to cause issues in 0.21)

komoricodrutz, 2024-03-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi,
This seems to be happening for quite a while now. I noticed this months ago.
A .FCStd file created in the **Daily** version seems to not open correctly in the **Stable** version.
Getting lots of errors in the Report view and the "diskette" symbol shows on every sketch or datum plane.
[image]
As you can see in the screenshot, existing datum planes and sketches show that "diskette" symbol in their icons.
Sketch032 from the screenshot has been created now, when opening the file and it shows the normal sketch icon

### Full version info

```shell
<details>
<summary>Stable version Details</summary>

OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * BOLTSFC 2022.11.5
  * A2plus 0.4.60n
  * Lithophane
  * Assembly4 0.50.2
  * Glass
  * fasteners 0.4.62
  * fcgear 1.0.0

```

</details>

<details>
<summary>Daily version Build Details</summary>

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36617 (Git)
Build type: Release
Branch: main
Hash: 4845d391ed2702b0b51a9da905066485eb5ef7bc
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * BOLTSFC 2022.11.5
  * A2plus 0.4.60n
  * Lithophane
  * Assembly4 0.50.2
  * Glass
  * fasteners 0.4.62
  * fcgear 1.0.0

```

</details>
```

### Subproject(s) affected?

None

### Anything else?

Dump from the Report view when opening the file in the Stable version:
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:18  No extension found of type 'App::SuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'
12:22:19  No extension found of type 'Gui::ViewProviderSuppressibleExtension'

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2024-03-27:
Duplicate #12894

Roy-043, 2024-03-27:
This is the opposite of backward compatibility: this is [forward compatibility](https://en.wikipedia.org/wiki/Forward_compatibility).

Closing this as a duplicate.

closed 2024-03-27 by Roy-043