FreeCAD/FreeCAD #13182: AppImage crashes on older thinkpad GPU, while Archlinux package does not.

Harvie, 2024-03-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD-0.21.2-Linux-x86_64.AppImage crashes:

```
FreeCAD 0.21.2, Libs: 0.21.2R33771 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: undefined symbol: amdgpu_va_get_start_addr (search paths /usr/lib/dri, suffix _dri)
failed to load driver: swrast
WebEngineContext used before QtWebEngine::initialize() or OpenGL context creation failed.
QGLXContext: Failed to create dummy context
Failed to create OpenGL context for format QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples 0, swapBehavior QSurfaceFormat::DefaultSwapBehavior, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile) 
/tmp/.mount_FreeCAtk8R2a/AppRun: line 43: 132945 Aborted                 (core dumped) ${MAIN} "$@"
```

Frecad archlinux package 0.21.2-6 does not:

```
FreeCAD 0.21.2, Libs: 0.21.2R33771 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##
```

I am guessing this probably only affects older intel integrated GPU like the one in my thinkpad.

### Full version info

```shell
Broken version was downloaded from https://github.com/FreeCAD/FreeCAD/releases/download/0.21.2/FreeCAD-0.21.2-Linux-x86_64.AppImage
FreeCAD 0.21.2 Revision: 33771 (Git)
962abe530f3b7f573b8341cbc15de676  FreeCAD-0.21.2-Linux-x86_64.AppImage
13eecbb23c60948b41d7270b75370794576e4b0ccde302a6f3472f46e996fce6  FreeCAD-0.21.2-Linux-x86_64.AppImage
I cannot get more info from dialog, since GUI does not start up

This is the archlinux version which works flawlessly:

[code]
OS: Arch Linux (XFCE/Xfce Session)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: Czech/Czech Republic (cs_CZ)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ckuethe, 2024-03-28:
I can repro this and my workaround is to preload `libdrm_amdgpu.so`

`LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libdrm_amdgpu.so ./Freecad*.AppImage` which makes the appimage work for me on Ubuntu 22.04 (well, Pop! OS)

Harvie, 2024-03-28:
On archlinux this helps: `LD_PRELOAD=/usr/lib/libdrm_amdgpu.so ./FreeCAD-0.21.2-Linux-x86_64.AppImage`
So probably the library bundled in appimage is corrupt?

jeedaiian, 2024-04-26:
I am also facing a similar issue.
>MESA-LOADER: failed to open radeonsi: /usr/lib/dri/radeonsi_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
failed to load driver: radeonsi
MESA-LOADER: failed to open radeonsi: /usr/lib/dri/radeonsi_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
failed to load driver: radeonsi
MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
A3 available
WebEngineContext used before QtWebEngine::initialize() or OpenGL context creation failed.
QGLXContext: Failed to create dummy context
Failed to create OpenGL context for format QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples 0, swapBehavior QSurfaceFormat::DefaultSwapBehavior, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile) 
/tmp/.mount_FreeCARzU7Fn/AppRun: line 43: 137683 Aborted                 (core dumped) ${MAIN} "$@"

glxinfo:
>Vendor: AMD (0x1002)
    Device: AMD Radeon Graphics (radeonsi, renoir, LLVM 15.0.7, DRM 3.57, 6.8.0-76060800daily20240311-generic) (0x1636)

This issue is not present in the FreeCAD weekly build AppImage, FreeCAD_weekly-builds-36277-2024-03-04-conda-Linux-x86_64-py310_xx

RossBarnie, 2024-05-16:
Can confirm it's fixed for me in `FreeCAD_weekly-builds-37302-conda-Linux-x86_64-py311.AppImage`.

luzpaz, 2024-08-14:
@Harvie :point_up_2: can you confirm ?

closed 2024-08-14 by luzpaz