FreeCAD/FreeCAD #13185: the 'freecad.gear' addon crashes freecad outright in the latest Dev build (and numerous prior dev builds too)

Malendryn, 2024-03-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I just downloaded the nightly build, (36596) with a fresh install,  
opened a new 'empty file', 
installed the 'freecad.gears' addon and restarted, then opened a new 'empty file' and tried to create an involute gear.
-- the program simply 'went away'  crashed with no message or popup whatsoever.

Tried again, from the command line using freecad -l and did the exact same thing.  this is what it returned on crashing:

`Fatal Python error: PyThreadState_Get: the function must be called with the GIL held, but the GIL is released (the current Python thread state is NULL)
Python runtime state: initialized

Current thread 0x00003874 (most recent call first):
  File "C:\Users\malen\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\pygears\_functions.py", line 83 in _func
  File "C:\Users\malen\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\pygears\involute_tooth.py", line 130 in undercut_points
  File "C:\Users\malen\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\pygears\involute_tooth.py", line 144 in points
  File "C:\Users\malen\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\involutegear.py", line 297 in generate_gear_shape
  File "C:\Users\malen\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 127 in execute
  File "C:\Users\malen\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\commands.py", line 63 in Activated

Extension modules: FreeCAD, FreeCADGui, pivy._coin, shiboken2.shiboken2, PySide2.QtCore, PySide2.QtGui, PySide2.QtWidgets, PySide2.QtNetwork, WebGui, StartGui, Start, Part, Sketcher, _PartDesign, numpy.core._multiarray_umath, numpy.core._multiarray_tests, numpy.linalg._umath_linalg, numpy.fft._pocketfft_internal, numpy.random._common, numpy.random.bit_generator, numpy.random._bounded_integers, numpy.random._mt19937, numpy.random.mtrand, numpy.random._philox, numpy.random._pcg64, numpy.random._sfc64, numpy.random._generator, PartGui, SketcherGui, PartDesignGui, scipy._lib._ccallback_c (total: 31)
Abnormal program termination...
Terminating...
`

_This has been reported as fixed numerous times, and yet keeps cropping up as a bug periodically.  (For me, the last time this worked reliably was version 0.21.2 which is the version I'm still using while waiting for this bug to be resolved)_

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36596 (Git)
Build type: Release
Branch: main
Hash: 380d452adb5da85f47543a316b6b533f84e829c7
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * freecad.gears 1.2.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-28:
Can someone confirm?
Is this an issue which needs to be resolved in the addon https://github.com/looooo/freecad.gears or is this related to the code in main
@looooo FYI

Syres916, 2024-05-03:
The line of code triggering the crash is a translation and the version of OCCT is already known to cause translation issues, see https://github.com/FreeCAD/FreeCAD/issues/9651
Therefore this can be linked to #9651 and closed as a duplicate IMHO. The build of 0.21 that the user is falling back on probably uses OCC 7.6.3, hence it works fine.

maxwxyz, 2024-05-03:
duplicate of https://github.com/FreeCAD/FreeCAD/issues/9651