FreeCAD/FreeCAD #13189: [Problem] Sketcher_Translate: Horizontal and vertical constraints are lost

DrD4ffy, 2024-03-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating copies or moving geometry with Sketcher_Translate, horizontal and vertical constraints are not preserved.

1. Open s sketch
2. Create a rectangle
3. Select the complete rectangle
4. Move (0 copies) or create a copy with Sketcher_Translate
5. The new geometry will no longer have horizontal and vertical constraints

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36596 (Git)
Build type: Release
Branch: main
Hash: 380d452adb5da85f47543a316b6b533f84e829c7
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Assembly4 0.50.9
  * Curves 0.6.24
  * DynamicData 2.60.0
  * ExplodedAssembly
  * fasteners 0.5.15
  * Freecad-Built-in-themes-beta 1.2.2
  * IconThemes
  * Ondsel-Lens 2024.1.22.01
  * OpenDark 2023.12.17
  * OpenLight 2024.2.7
  * PieMenu 1.3.9
  * Pyramids-and-Polyhedrons
  * QuickMeasure 2022.10.28
  * Render 2024.1.23
  * SelectorToolbar
  * sheetmetal 0.4.6
  * ThreadProfile 1.89.0
```

### Subproject(s) affected?

Sketcher

### Anything else?

Perhaps a 'copy/paste' behavior from the Sketcher_Rotate tool, where hor/vert constraints cannot be preserved. But I think Sketcher_Translate should.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-28:
did you check the clone constraints option or not?
@PaddleStroke FYI

PaddleStroke, 2024-03-28:
I can reproduce the bug. It should work even without clone constraint option (which only replace dimensions with equalities between copies)

kaiwas, 2024-03-28:
@DrD4ffy  This feature is only part of a larger problem. With the help of transformation tools, not only vertical and horizontal constraints are lost
https://github.com/FreeCAD/FreeCAD/issues/13013

PR #13190 (2024-04-05): Sketcher: Translate Fixes #13189 horizontal/vertical lost.
Fixes #13189
Sketcher_Translate: Horizontal and vertical constraints are lost
commit 5efc44d5 (2024-04-05): Sketcher: Translate Fixes #13189 horizontal/vertical lost.
closed 2024-04-05 by chennes