FreeCAD/FreeCAD #13193: [Suggestion] Creating a Group should be insert into the currently selected Group

willrabbermann, 2024-03-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating a new Std_Group with another Std_Group currently selected, the new Group should be placed into selected Group. Currently it is inserted into the very bottom of the Document tree. This makes working on a very large project more cumbersome than it should need to be. 

Currently you have to scroll all the way to the end of the document to find the new Std_Group object every-time you want to reorganize some objects.

### Full version info

```shell
0.21.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-28:
As far as I know, you cannot have a selection and then create a group?

willrabbermann, 2024-03-28:
Std_group is inferior to Std_Part. All groups can do is toggle visibility of its members. Currently, group addition with group selection will function as I stated.

FEA-eng, 2024-08-09:
I wanted to request something more general - being able to activate or select a Group like a Body or Part container so that all new objects are added to it as long as it's activated/selected. This would eliminate the need to manually drag all objects to the Group container. 

I'm not sure if it should be a separate issue or part of this one.