FreeCAD/FreeCAD #13194: The context-menu of the tree view is too overloaded

wwmayer, 2024-03-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

For a selected object the context-menu of the tree view contains too many commands. The issue was brought up [here](https://github.com/FreeCAD/FreeCAD/pull/12994)

E.g, for a selected part object it currently contains 28 top-level commands plus extra commands of some sub-menus.
[image]

A context-menu should only contain commands that relate to the selected object and not any unrelated commands. And it should only contain the more important commands while for less important commands it's sufficient to access them via the menu bar.

* From the context-menu shown above I would remove the commands: Cut, Copy, Paste and Properties
* Tree settings should only be shown if nothing is selected as this only relates to the tree view
* The commands Show selection, Hide selection and Toggle selectability are at least questionable

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36642 (Git)
Build type: debug
Branch: main
Hash: 1eea88a8b72f3104e3756474d4c3df8c22a6e0f9
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-28:
related: https://github.com/FreeCAD/FreeCAD/issues/12826

X-Ryl669, 2024-03-28:
In this menu, you have 5 toggle commands, many of them are redundant (like visibility) and incompatible or logically impossible (once an item is invisible, you can't select it because it doesn't exist as an visual object anymore).

Some are clearly useless in a menu (like "Toggle transparency" which could be done in the Appearance dialog, or to avoid opening a dialog, in a submenu).

Why is "Set color" not in the "Appearance" section ?
What is the use of "Random color" ? Has anyone used that in the last month, last year ? Are we playing a game here? 

Also, there's the Fitz Law issue, the most used command should be as close as possible to the cursor. With the upcoming toponaming feature, you're more likely to select "Recompute". It must be first or in the first items.

A circular menu (like FormIt, or Fusion360) is more convenient and you can group related items in submenu for "advanced" settings. If it's not possible to make a circular menu, then an "context toolbar" menu (à la Solidworks, where only icons are show on a toolbar that appears close to your right clicked cursor) also solves Fitz Law problem.

However, an icon only solution works if the icons are different enough.  I know I won't make friend criticizing one of the biggest issue in FreeCAD: the exponential icons problem. Every workspace brings in its numerous icons in the toolbars, making them unusable, hard to spot, and sometimes, most of the time, useless.

wwmayer, 2024-03-28:
> Why is "Set color" not in the "Appearance" section ?

This is part of the Part wb while the others are from the core system and added afterwards. So, will be quite difficult to change the order.

> What is the use of "Random color" ? Has anyone used that in the last month, last year ? Are we playing a game here?

I used it recently. But yes it can be removed from the context-menu as it's too much of a niche function.

> With the upcoming toponaming feature, you're more likely to select "Recompute". It must be first or in the first items.

The top position must always be reserved for the most suitable command that you trigger with a double click. The text of the command is bold.

X-Ryl669, 2024-03-28:
> This is part of the Part wb while the others are from the core system and added afterwards. So, will be quite difficult to change the order.

Why ?

> The top position must always be reserved for the most suitable command that you trigger with a double click. The text of the command is bold.

Why ? 

I'm not trying to troll here. Nor can I be exhaustive with all the menu issues. I'm just a user that used to work with different CAD suites and I find FreeCAD to be subpar with all the other software. Not in the feature set, but in the UX mainly.

Other CAD software don't have the "default double click action" takes one menu row (because, well, it's useless one since one can double click). They don't care about what order the items were added in the menu and they don't have 1 thousand choices in the menu. If you need so many choices, just select the item and use the global menu bar. It's actually easier to select an action in the global menu bar than in a context menu (simply because you'll quickly move your cursor up to the top of the screen, the cursor will stop on top of the screen anyway, you don't have to be precise) and then you can select your action. A context menu requires precise pointing, and, in some case, scrolling.

I think the menu should go through some kind of runtime filter to remove items or reorder them if it can't be done at design time. Its current state is anarchic and it's becoming worse with #12994. Ideally, the menu shouldn't be displayed like it is.

Maybe adding a statistic collector for some release versions to collect usage statistic for each menu selection (sorted by current active workbench, current active item) and then remove (or move to some submenu) the unused items.

I've seen they are some work on trying to remove the pesky report window to have something more usable from a user perspective, but the menus and toolbars need the same love if not more.

X-Ryl669, 2024-03-28:
For reference here are the other context menu in other CAD suite:
Solidworks (with a context icon box with most used actions):
[image]
[image]

Fusion360 (circular menu and context menu):
[image]

Formit (circular menu):
[image]

Blender (circular menu, notice the consistent shortcuts):
[image]
And all its possible context menu:
[image]

marcin86junior, 2024-04-02:
Maybe new sub-menu:

[image]

wwmayer, 2024-04-14:
* https://forum.freecad.org/viewtopic.php?t=86891
* https://forum.freecad.org/viewtopic.php?t=86892

maxwxyz, 2026-11-07:
Part of this is implemented in Werner’s fork: https://codeberg.org/wwmayer/FreeCAD11/commit/2377976b61a1209f13b369b795f12b2241c3d8c0

It removes several clipboard and unrelated entries from the tree menu and only shows Tree settings without a selection. It keeps some appearance commands, so it is not the whole redesign discussed here.