FreeCAD/FreeCAD #13198: [Suggestion] Make Draft_Dimension "Distance" value to system clipboard

willrabbermann, 2024-03-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I notice FreeCad will not allow user to copy the "Distance" value directly to the clipboard. User must find the value by selecting the dimension and type it in somewhere else or make the font bigger which is slow.

Lets make that value copy-able to the clipboard. 
[image]

### Full version info

```shell
0.21.2
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-28:
To add, the dimension value could be displayed in the label to find the right dimensions quicker

benj5378, 2024-10-05:
This is for all disabled property entries in FreeCAD

benj5378, 2024-10-05:
Given the following backtrace, that was a forever needlestick operation, I believe the issue is due to the filtering in `QWidget::event`. All key press events guarded if the widget is disabled. We therefore have to set up our `own` event function.

```cpp
#0  0x00007ffff1959070 in QGuiApplication::clipboard()@plt () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#1  0x00007ffff1abd8b5 in QWidgetLineControl::copy(QClipboard::Mode) const (this=this@entry=0x555558311890, mode=mode@entry=QClipboard::Clipboard)
    at widgets/qwidgetlinecontrol.cpp:157
#2  0x00007ffff1ac2e2e in QWidgetLineControl::processKeyEvent(QKeyEvent*) (this=0x555558311890, event=event@entry=0x7fffffffc950)
    at widgets/qwidgetlinecontrol.cpp:1736
#3  0x00007ffff1ab5ad3 in QLineEdit::keyPressEvent(QKeyEvent*) (this=0x5555597b7260, event=0x7fffffffc950) at widgets/qlineedit.cpp:1793
#4  0x00007ffff19afa2a in QWidget::event(QEvent*) (this=0x5555597b7260, event=0x7fffffffc950) at kernel/qwidget.cpp:8703
#5  0x00007ffff1b17dae in QAbstractSpinBox::keyPressEvent(QKeyEvent*) (this=0x555559e1ab20, event=0x7fffffffc950)
    at widgets/qabstractspinbox.cpp:1140
#6  0x00007ffff60472a6 in Gui::QuantitySpinBox::keyPressEvent(QKeyEvent*) (this=0x555559e1ab20, event=0x7fffffffc950)
    at /home/bensay/Development/FreeCAD/src/Gui/QuantitySpinBox.cpp:411
#7  0x00007ffff19afa2a in QWidget::event(QEvent*) (this=0x555559e1ab20, event=0x7fffffffc950) at kernel/qwidget.cpp:8703
#8  0x00007ffff60493cf in Gui::QuantitySpinBox::event(QEvent*) (this=0x555559e1ab20, event=0x7fffffffc950)
    at /home/bensay/Development/FreeCAD/src/Gui/QuantitySpinBox.cpp:878
#9  0x00007ffff196bf92 in QApplicationPrivate::notify_helper(QObject*, QEvent*)
    (this=this@entry=0x5555556e8600, receiver=receiver@entry=0x555559e1ab20, e=e@entry=0x7fffffffc950) at kernel/qapplication.cpp:3640
#10 0x00007ffff1973f97 in QApplication::notify(QObject*, QEvent*) (this=<optimized out>, receiver=0x555559e1ab20, e=<optimized out>)
    at kernel/qapplication.cpp:3002
#11 0x00007ffff5914dac in Gui::GUIApplication::notify(QObject*, QEvent*) (this=0x7fffffffd710, receiver=0x555559e1ab20, event=0x7fffffffc950)
    at /home/bensay/Development/FreeCAD/src/Gui/GuiApplication.cpp:93
#12 0x00007ffff0cbae38 in QCoreApplication::notifyInternal2(QObject*, QEvent*) (receiver=0x555559e1ab20, event=0x7fffffffc950)
    at kernel/qcoreapplication.cpp:1064
#13 0x00007ffff0cbae86 in QCoreApplication::forwardEvent(QObject*, QEvent*, QEvent*)
```

 After hours of debugging, that focus might be the reason

https://github.com/qt/qtbase/blob/4dc56a7e4329e5e90ae769b52da5350f15b3c8ab/src/widgets/kernel/qwidget.cpp#L3395-L3399

Now I have succesfully checked, that the object has focus

```cpp
Thread 1 "FreeCAD" hit Breakpoint 1.17, QWidget::setFocus (this=0x5555582ace20, reason=Qt::OtherFocusReason) at kernel/qwidget.cpp:6296
6296	//{
#0  QWidget::setFocus(Qt::FocusReason) (this=0x5555582ace20, reason=Qt::OtherFocusReason) at kernel/qwidget.cpp:6296
#1  0x00007ffff58ee10f in QWidget::setFocus() (this=0x5555582ace20) at /usr/include/x86_64-linux-gnu/qt5/QtWidgets/qwidget.h:420
#2  0x00007ffff5e06e15 in Gui::PropertyEditor::PropertyItemDelegate::createEditor(QWidget*, QStyleOptionViewItem const&, QModelIndex const&) const
    (this=0x5555560cc840, parent=0x5555560cd870, index=...) at /home/bensay/Development/FreeCAD/src/Gui/propertyeditor/PropertyItemDelegate.cpp:196
PropertyEditor::eventFilter
```

And still, it receives no events at all, solely because it's disabled (making it enabled at creation, then it receives event). I have checked many times, that we don't skip elements that are disabled... 

And it should be possible receiving events even though disabled. I even tried setting up an event filter.

It might seem however, that the event filter receives mouse event, though no key events for some very weird reason, even though it has focus.

Having spent 6 hours on this, I will give up for now.