FreeCAD/FreeCAD #13201: [Suggestion] Add Annotation editor current line color to colors settings

willrabbermann, 2024-03-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I can barely read the annotation current line. As far as I know its color setting is not located in Preferences->Tech Draw->Colors.

### Full version info

```shell
0.21.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-28:
Please provide full about info.
Are you using a theme?

willrabbermann, 2024-03-28:
Default theme. Is that where its set?

maxwxyz, 2024-03-28:
@MisterMakerNL FYI

MisterMakerNL, 2024-03-28:
This is an oldy, I think we even updated the text input window to get this
fixed. But I'm 95% sure it's fixed in 0.22.

https://ci3.googleusercontent.com/meips/ADKq_NbUp1hF1rxxu0ypqckEzrGnRN5W6AkA4WaBbfDF1jTXjfdN2TRNhQ3w0af57TdJfnLP8WZfOOq8XXI5X404F_4WqfQbim_2pvNZz4_ebJOYBWrPaVIeL7gLbo0hv0XsFfXQU9B7o3DVHoR5kCL0A63bRVt2TxKnm0LhRPCXYd788egvdXYi5hWBn1Lnu1OJ7gd8hvAsYmxpL6-8txrr8MR0_bLPn2Sb1L-kKlhiOJKN0Q-yynN_WeE=s0-d-e1-ft#https://github.com/notifications/beacon/AHDMTIVIDJGB6LZQY4ZPIRLY2RA2LA5CNFSM6AAAAABFM74F2SWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTTYX2RFO.gif>Message
ID: ***@***.***>

maxwxyz, 2024-03-29:
@willrabbermann could you please try this in the current weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

maxwxyz, 2024-03-29:
Already implemented

closed 2024-03-29 by maxwxyz

willrabbermann, 2024-03-29:
[image]
Should I be using the rich text editor instead, its editor looks better.

maxwxyz, 2024-03-29:
Depends on your use case.

willrabbermann, 2024-03-29:
The problem is still under the ProDark theme. That was the one I had selected.
Other themes appear fine.
[image]
[image]