FreeCAD/FreeCAD #13203: TechDraw: Disable hatching for individual shapes

maxwxyz, 2024-03-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently it is not possible to disable / remove the hatching of selected features (e.g. a body, a shape,..). When a section view is created, either all or nothing is displayed with hatching. It should be possible for section cuts of assemblies or parts, to select closed loops of shapes or bodies and delete the hatching for that region. Like it it possible to add custom hatching to a face (maybe kind of adding a blank hatching without a pattern). Use case, e.g. when cutting bearings, the balls are never displayed with a hatching.
 

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36487 (Git)
Build type: Release
Branch: main
Hash: f5baef888c25937026ee50001a4fe4293a9f11ee
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.3
  * CurvedShapes 1.0.8
  * Curves 0.6.29
  * Defeaturing 1.2.2
  * fasteners 0.5.15
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.6
  * sheetmetal.backup1710961794.9712274 0.4.3 (Disabled)
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-04-04:
Right, excluding objects/parts from hatching would be very useful since some features shouldn't be hatched (like ribs and fasteners).

https://forum.freecad.org/viewtopic.php?t=68924

ximeg, 2024-04-17:
I second this: there has to be a way to remove the previously applied hatch. I sometimes modify the original geometry, which leads to recomputing the drawing and assigning the hatches to the wrong faces. If I can't remove that hatch, I have to recreate the entire view in the drawing.

Here is an example where I changed one of the model dimensions.
Cross-section before the change
[image]

Cross-section after the change
[image]

WandererFan, 2024-05-18:
JFYI, if you set the section face to no hatch, then you can access the individual faces to assign hatches to each.

maxwxyz, 2024-05-18:
thanks for mentioning this workaround, although not ideal if I only want one face not hatched.