FreeCAD/FreeCAD #13204: TechDraw: Exclude shapes / bodies from section view

maxwxyz, 2024-03-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently it is not possible to remove / exclude bodies, parts or shapes from a section view. The feature cannot be in the parent / reference view where the section view is based on.
It should be possible to select bodies, parts or shapes either in the 3D view or on the 2D drawing and exclude them from the cut to be displayed without respect of the section line.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36487 (Git)
Build type: Release
Branch: main
Hash: f5baef888c25937026ee50001a4fe4293a9f11ee
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.3
  * CurvedShapes 1.0.8
  * Curves 0.6.29
  * Defeaturing 1.2.2
  * fasteners 0.5.15
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.6
  * sheetmetal.backup1710961794.9712274 0.4.3 (Disabled)
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-04-04:
So you mean excluding from hatching not from being shown in the section view? Excluding from hatching would indeed be very useful since some features shouldn't be hatched (like ribs and fasteners).

https://forum.freecad.org/viewtopic.php?t=68924

WandererFan, 2024-05-18:
I don't understand this one.   If the base view has source objects A,B,C,D, you want the section to only cut B & C, but not A & D?  And A & D should be in the section view as uncut?

maxwxyz, 2024-05-18:
@WandererFan yes

FEA-eng, 2024-05-18:
> @WandererFan yes usually standard parts like screws, keyways, but also other things like teeth of gears,... 

... or ribs as also shown in your screenshot and discussed here: https://github.com/FreeCAD/FreeCAD/issues/13203#issuecomment-2036468788

According to the classic Polish book about technical drawings (by Dobrzanski), the following items shouldn't be hatched:
- sections passing along objects' walls, ribs and spokes
- solid objects of revolution such as pins, rivets and shafts whose axes are in the plane of the cut
- other solid or hollow machine parts whose shapes are obvious - screws, nuts, wedges, keys

WandererFan, 2024-12-30:
another forum post on the topic: https://forum.freecad.org/viewtopic.php?t=93368