FreeCAD/FreeCAD #13206: CAM should read Path preferences on first start

chrisb-github, 2024-03-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On a non developer user machine there is usually one FreeCAD installed. An upgrade from 0.21 to 1.0 (or 0.22), makes it difficult to get the Path preferences. That aren't too many, but to make sure, it would need after replacing 0.21 with 1.0:
- find out how to install multiple versions
- find out where to download the old version
- download 0.21, which is bandwidth intensive
- unpack
- start both FreeCADs
- compare the settings

None of these is something that couldn't be handled, but it's a hassle.

Proposal:
If CAM starts, it checks if there are cam settings. If not it moves Path settings to CAM settings.
This code can be removed in future versions.

### Full version info

```shell
OS: macOS 13.0
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36391 (Git)
Build type: Release
Branch: main
Hash: 52fffaf50ef8facea2ecfde23d6f1a7d3ad3d80c
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * FC_SU
  * freecad.gears 1.2.0
  * FeedsAndSpeeds 0.5.0
  * dxf-library
  * fasteners 0.5.13
  * sheetmetal 0.4.2
  * ExplodedAssembly
  * Curves 0.6.24
```

### Subproject(s) affected?

CAM/Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-28:
@sliptonic FYI

closed 2024-04-02 by sliptonic

wolfgangr, 2024-05-02:
CAM still does not migrate Path settings in 36731 (2024-04-02) and 37083 (last weekly as of 2027-05-03)
see detailled tests as instructed by @chrisb-github chrisb here:
https://forum.freecad.org/viewtopic.php?p=757001#p757001

maxwxyz, 2024-05-03:
@sliptonic FYI

commit 5cd958de (2024-04-02): fixes #13206