FreeCAD/FreeCAD #13210: FreeCAD current don't compile with OCCT-7.8.1 with recent add WireJoiner.cpp

Tarot67, 2024-03-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There are a issue when build freecad (master) branch:

[ 23%] Building CXX object src/Mod/Part/App/CMakeFiles/Part.dir/WireJoiner.cpp.o
In file included from /compile/freecad-test/src/Mod/Part/App/WireJoiner.cpp:64:
/compile/freecad-test/src/Mod/Part/App/WireJoiner.cpp: In member function ‘bool Part::WireJoiner::WireJoinerP::addValidEdges(const TopoDS_Edge&, gp_Pnt, double, TopoDS_Vertex&, TopoDS_Edge&, gp_Pnt, TopoDS_Vertex&, TopoDS_Edge&, bool)’:
/compile/freecad-test/src/Mod/Part/App/WireJoiner.cpp:789:52: error: ‘const class TopoDS_Edge’ has no member named ‘HashCode’
  789 |                 FC_MSG("addcheck " << vinfo.edge().HashCode(INT_MAX));
      |                                                    ^~~~~~~~
/compile/freecad-test/src/Base/Console.h:364:53: note: in definition of macro ‘__FC_PRINT’
  364 |             _instance.prefix(_str, _file, _line) << _msg;                                          \
      |                                                     ^~~~
/compile/freecad-test/src/Base/Console.h:376:22: note: in expansion of macro ‘_FC_PRINT’
  376 | #define FC_MSG(_msg) _FC_PRINT(FC_LOG_INSTANCE, FC_LOGLEVEL_MSG, Message, _msg)
      |                      ^~~~~~~~~
/compile/freecad-test/src/Mod/Part/App/WireJoiner.cpp:789:17: note: in expansion of macro ‘FC_MSG’
  789 |                 FC_MSG("addcheck " << vinfo.edge().HashCode(INT_MAX));
      |                 ^~~~~~
/compile/freecad-test/src/Mod/Part/App/WireJoiner.cpp: In member function ‘void Part::WireJoiner::WireJoinerP::printHistoryInit(const Handle_BRepTools_History&, const std::vector<Part::TopoShape>&)’:
/compile/freecad-test/src/Mod/Part/App/WireJoiner.cpp:2715:80: error: ‘const class TopoDS_Shape’ has no member named ‘HashCode’
 2715 |             FC_MSG(shape.getShape().TShape().get() << ", " << shape.getShape().HashCode(INT_MAX));
      |                                                                                ^~~~~~~~
/compile/freecad-test/src/Base/Console.h:364:53: note: in definition of macro ‘__FC_PRINT’
  364 |             _instance.prefix(_str, _file, _line) << _msg;                                          \
      |                                                     ^~~~
/compile/freecad-test/src/Base/Console.h:376:22: note: in expansion of macro ‘_FC_PRINT’
  376 | #define FC_MSG(_msg) _FC_PRINT(FC_LOG_INSTANCE, FC_LOGLEVEL_MSG, Message, _msg)
      |                      ^~~~~~~~~
/compile/freecad-test/src/Mod/Part/App/WireJoiner.cpp:2715:13: note: in expansion of macro ‘FC_MSG’
 2715 |             FC_MSG(shape.getShape().TShape().get() << ", " << shape.getShape().HashCode(INT_MAX));
      |             ^~~~~~
/compile/freecad-test/src/Mod/Part/App/WireJoiner.cpp: In member function ‘void Part::WireJoiner::WireJoinerP::printHistoryFinal()’:
/compile/freecad-test/src/Mod/Part/App/WireJoiner.cpp:2729:58: error: ‘class TopoDS_Edge’ has no member named ‘HashCode’
 2729 |             FC_MSG(shape.TShape().get() << ", " << shape.HashCode(INT_MAX));
      |                                                          ^~~~~~~~
/compile/freecad-test/src/Base/Console.h:364:53: note: in definition of macro ‘__FC_PRINT’
  364 |             _instance.prefix(_str, _file, _line) << _msg;                                          \
      |                                                     ^~~~
/compile/freecad-test/src/Base/Console.h:376:22: note: in expansion of macro ‘_FC_PRINT’
  376 | #define FC_MSG(_msg) _FC_PRINT(FC_LOG_INSTANCE, FC_LOGLEVEL_MSG, Message, _msg)
      |                      ^~~~~~~~~
/compile/freecad-test/src/Mod/Part/App/WireJoiner.cpp:2729:13: note: in expansion of macro ‘FC_MSG’
 2729 |             FC_MSG(shape.TShape().get() << ", " << shape.HashCode(INT_MAX));
      |             ^~~~~~
/compile/freecad-test/src/Mod/Part/App/WireJoiner.cpp: In member function ‘void Part::WireJoiner::WireJoinerP::printHistoryOfShape(const opencascade::handle<BRepTools_History>&, const T&)’:
/compile/freecad-test/src/Mod/Part/App/WireJoiner.cpp:2791:71: error: ‘class TopoDS_Shape’ has no member named ‘HashCode’
 2791 |                    << it.Value().TShape().get() << ", " << it.Value().HashCode(INT_MAX));
      |                                                                       ^~~~~~~~
/compile/freecad-test/src/Base/Console.h:364:53: note: in definition of macro ‘__FC_PRINT’
  364 |             _instance.prefix(_str, _file, _line) << _msg;                                          \
      |                                                     ^~~~
/compile/freecad-test/src/Base/Console.h:376:22: note: in expansion of macro ‘_FC_PRINT’
  376 | #define FC_MSG(_msg) _FC_PRINT(FC_LOG_INSTANCE, FC_LOGLEVEL_MSG, Message, _msg)
      |                      ^~~~~~~~~
/compile/freecad-test/src/Mod/Part/App/WireJoiner.cpp:2789:13: note: in expansion of macro ‘FC_MSG’
 2789 |             FC_MSG(shape.getShape().TShape().get()
      |             ^~~~~~
/compile/freecad-test/src/Mod/Part/App/WireJoiner.cpp: In instantiation of ‘void Part::WireJoiner::WireJoinerP::printHistoryOfShape(const opencascade::handle<BRepTools_History>&, const T&) [with T = Part::TopoShape]’:
/compile/freecad-test/src/Mod/Part/App/WireJoiner.cpp:2800:32:   required from ‘void Part::WireJoiner::WireJoinerP::printHistory(opencascade::handle<BRepTools_History>, const T&) [with T = std::unordered_set<Part::TopoShape, Part::ShapeHasher, Part::ShapeHasher>]’
/compile/freecad-test/src/Mod/Part/App/WireJoiner.cpp:2717:21:   required from here
/compile/freecad-test/src/Mod/Part/App/WireJoiner.cpp:2790:48: error: ‘const class TopoDS_Shape’ has no member named ‘HashCode’
 2790 |                    << ", " << shape.getShape().HashCode(INT_MAX) << " -> "
      |                               ~~~~~~~~~~~~~~~~~^~~~~~~~
/compile/freecad-test/src/Base/Console.h:364:53: note: in definition of macro ‘__FC_PRINT’
  364 |             _instance.prefix(_str, _file, _line) << _msg;                                          \
      |                                                     ^~~~
/compile/freecad-test/src/Base/Console.h:376:22: note: in expansion of macro ‘_FC_PRINT’
  376 | #define FC_MSG(_msg) _FC_PRINT(FC_LOG_INSTANCE, FC_LOGLEVEL_MSG, Message, _msg)
      |                      ^~~~~~~~~
/compile/freecad-test/src/Mod/Part/App/WireJoiner.cpp:2789:13: note: in expansion of macro ‘FC_MSG’
 2789 |             FC_MSG(shape.getShape().TShape().get()
      |             ^~~~~~
make[2]: *** [src/Mod/Part/App/CMakeFiles/Part.dir/build.make:3423: src/Mod/Part/App/CMakeFiles/Part.dir/WireJoiner.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:7050: src/Mod/Part/App/CMakeFiles/Part.dir/all] Error 2
make: *** [Makefile:146: all] Error 2

### Full version info

```shell
I am in branch master because i test Freecad dev with Qt6

bash-5.2$ git status
On branch main
Your branch is up to date with 'origin/main'.
```

### Subproject(s) affected?

None

### Anything else?

In past i read some error like this in some commit!. But i not able to resolve ;.( himself

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-29:
@CalligaroV FYI

CalligaroV, 2024-03-29:
Hi @Tarot67 and @maxwxyz, thanks for reporting!

I confirm the issue, it is caused by the rework of the hashing logic from OCCT 7.8.0:
https://dev.opencascade.org/doc/overview/html/occt__upgrade.html#upgrade_780_hash_utils

I applied a modification and tested again with both OCCT 7.8.0 and the version I usually use (7.6.3).
In both cases I can compile and the tests for `WireJoiner` still gives the expected result.

There's a PR for this issue, #13218, if you too can try it to see if it fixes the issue we'll have other confirmations that we can merge it.

Thanks!

Tarot67, 2024-03-29:
I confirm your new file WireJoin.cpp (patch with pragma) build now with OCCT-7.8.1.
Your had resolved issue one less than one day!. amazing. Thank you very much

Application running well

Well Done ;)

For me you can merge and close issue

jakeogh, 2024-03-30:
#13218 tested and working here.

CalligaroV, 2024-03-30:
Thank you very much @Tarot67 and @jakeogh for the feedback!

@chennes I would say we can merge and close this unless there are other modifications needed

closed 2024-04-02 by chennes