FreeCAD/FreeCAD #13211: FEM: when deleting a pipeline results, the results remain visible until reopen

MSoegtropIMC, 2024-03-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In FEM mode, when I delete a pipeline results object, the results remain visible and the only way to get rid of them I found is to save, close and reopen the file.

When I hide the pipeline results object before deleting it, there is no issue.

This is 100% reproducible in > 50 tests. As far as I can tell this happens always in all projects. I have never seen it not happening.

I am usually using Gmesh and Calculix.

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Germany (en_DE)
[/code]
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-29:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/11175

maxwxyz, 2024-03-29:
Closing as already implemented

closed 2024-03-29 by maxwxyz