FreeCAD/FreeCAD #13213: "Unhandled unknown exception caught in selection observer" when trying to apply a FEM constraint

FEA-eng, 2024-03-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a simple egg-shaped geometry in STEP format. 

[image]

[Egg.zip](https://github.com/FreeCAD/FreeCAD/files/14802489/Egg.zip)

When I try to select any of its 2 surfaces for a FEM constraint, I get the following error:

```
10:47:56  279.281 Selection.cpp(159): Unhandled unknown exception caught in selection observer
10:47:57  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'Base.FreeCADError'>: Unknown C++ exception
```

I've tried recreating this geometry in Link Stable (because there I can use the defining geometry in a sketch) and exporting it to STEP. Now it has only one face with a seam edge. But when I import it to master, I still get those errors in FEM. No idea what's so special about this geometry, it doesn't seem to be invalid in any way. The same thing happens in 0.21.2 so it's not a new bug.

@marioalexis84 FYI

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36596 (Git)
Build type: Release
Branch: main
Hash: 380d452adb5da85f47543a316b6b533f84e829c7
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-29:
how did you model the egg, do you have the FCStd? I'm not sure if this is only FEM related?

FEA-eng, 2024-03-29:
> how did you model the egg, do you have the FCStd? I'm not sure if this is only FEM related?

I modeled it in another software and can only export it to neutral format and recreate if needed (I've tried it 0.22 too - just downgraded in Draft and converted the wires to a sketch which I then revolved in PD - same errors occur). It seems to be FEM-specific because other operations, e.g. in Part or Part Design, work normally on this geometry.

maxwxyz, 2024-03-29:
I have no idea whats wrong, when downgrading and creating a sketch I can create a revolution but get the same error. When trying to change the sketch to a simple triangle I cannot add a revolution or even a pad after I've created an Analysis container. Just a bunch of errors

FEA-eng, 2024-03-29:
Yeah, it's really strange. I've even tried with a simple sphere exported to STEP but this one is working correctly. So either the egg shape is so specific and causes issues because of its curvature or the model is broken starting even from the sketch but it shouldn't be the case. I can use this model normally in all other ways and only can't apply FEM constraints to it.

maxwxyz, 2024-03-29:
Can you create another model and apply FEM constraints when it is based on a revolution of a sketch with a BSpline, created in FC?

FEA-eng, 2024-03-29:
> Can you create another model and apply FEM constraints when it is based on a revolution of a sketch with a BSpline, created in FC?

Good point. It turns out that I can't. I attached a FreeCAD file. But it's interesting that the same happens with STEP geometry where B-spline was used in another software.

Then the issue is more general and serious than I expected.

[B-spline revolved FEM.zip](https://github.com/FreeCAD/FreeCAD/files/14802928/B-spline.revolved.FEM.zip)

maxwxyz, 2024-03-29:
Can you rephrase it then? Seems like FEM constraints can just handle plane or rounded faces, no irregular surfaces?

FEA-eng, 2024-03-29:
> Seems like FEM constraints can just handle plane or rounded faces, no irregular surfaces?

Not exactly, padded B-spline works. Even with fillets. Maybe because there are more faces with less curvature. This is really strange. I'm surprised it has never been detected.

wwmayer, 2024-04-01:
https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Fem/App/FemConstraint.cpp#L433

A null vector cannot be normalized and OCC throws an exception that isn't handled anywhere inside the FEM module.

PR #13260 (2024-04-01): FEM: Handle unknown exception when applying a constraint to a B-Spli…
…ne surface

Fixes #13213

An OCC exception is raised when trying to normalize a null vector. The solution is to check the square length before normalizing it
commit 4b18a101 (2024-04-01): FEM: Handle unknown exception when applying a constraint to a B-Spline surface

Fixes #13213

An OCC exception is raised when trying to normalize a null vector. The solution is to check the square length before normalizing it
closed 2024-04-01 by wwmayer