FreeCAD/FreeCAD #13214: [Problem] PartDesign Revolution: automatic change of Reversed checkbox is confusing

Roy-043, 2024-03-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When changing the axis of a PD Revolution the Reversed checkbox is (un)checked automatically. This is confusing and does not always happen consistently.

[PD-revolve.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/14802543/PD-revolve.FCStd.zip)

1. Remove the .zip extension from the attached file and open it.
2. Activate the Body.
3. Select Sketch001.
4. Call PartDesign_Revolution.
5. Change the angle to 90 degrees.
6. Notice that the Reversed checkbox is checked.
7. Change the Axis to the Line object (the arrow indicates its direction).
8. Reversed checkbox is still checked.
9. Change the Axis to the Line001 object.
10. Reversed checkbox is still checked.
11. Change the Axis back to `Line:Edge1` by selecting that option from the dropdown.
12. Reversed checkbox is unchecked now.
13. Change the Axis back to `Vertical sketch axis`.
14. Reversed checkbox is checked again.

Apparently the goal is to have to +angle direction match the normal of the sketch. But why does PartDesign_Pad then not show a similar behavior? If you repeat the above scenario with Sketch002 and PartDesign_Pad the Reversed checkbox is never (un)checked automatically.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36596 (Git)
Build type: Release
Branch: main
Hash: 380d452adb5da85f47543a316b6b533f84e829c7
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

benj5378, 2024-10-05:
Could a viable solution be to remove the auto change of the reversed checkbox?

Roy-043, 2024-10-06:
That would work for me. But I do not have a full overview of the specifics of all PartDesign tools. Perhaps there are other tools that have this behavior that should also be checked/updated?

Roy-043, 2025-06-18:
Still an issue:
```
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.42149 (Git)
Build date: 2025/06/13 08:44:01
Build type: Release
Branch: main
Hash: 9714e2f87a6240d1b2b979eb39c7d34791aafdae
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Free Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

Roy-043, 2025-12-26:
Still an issue:
```
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.2.0dev.20251216 (Git shallow)
Build date: 2025/12/16 18:08:50
Build type: Release
Branch: (HEAD detached at 8fd81d4)
Hash: 8fd81d41094d90ebdb9b7e489c1e634884b7c4c1
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.4, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

Roy-043, 2025-12-27:
I am informed of 3 stale issues today. So it is not 1 per day.