FreeCAD/FreeCAD #13223: Make Datum plane visible during edit

dzid26, 2024-03-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Changing Datum plane visibility in the tree view is not possible if it is being edited. 

[image]
Currently, it is necessary to cancel the task, toggle the visibility (e.g. _space_ shortcut), and start the edit task again.  This is suboptimal UX. 

Some other features, if hidden, they automatically become visible when editing. 

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * 3D_Printing_Tools
  * CubeMenu
  * Defeaturing 1.2.2
  * fasteners 0.5.15
  * kicadStepUpMod 11.1.0
  * Launcher
  * ModernUI
  * Ondsel-Lens 2024.3.26.01
  * OpenDark 2023.12.17
  * PieMenu 1.4.2
  * POV-Ray-Rendering
  * Render 2024.1.23
  * ShortCuts
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-29:
It works. You are trying to change visibility of the current object you are currently editing. With other task dialogs all fine.

maxwxyz, 2024-03-29:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/11214

dzid26, 2024-03-29:
> Duplicate of #11214
I believe, this other issue is asking for more.

For the datum plane the editing preview functionality already exists - but it is not visible if the part was hidden. 

What we need is to switch datum plane visibility on, before the edit dialog window opens up.

dzid26, 2024-03-29:
This is actually more generic. and applies also to hidden bodies.  
E.g. try editing a feature of a hidden body. The edit window will open, but the body is invisible.  It is possible to toggle other bodies visibility, but not the one being edited.  Again, the one way to solve it would be to turn visibility _on_ automatically when before editing window opens up.

maxwxyz, 2024-03-29:
Can you rephrase the issue then?

dzid26, 2024-03-29:
I changed it the title and description. 

Do you think it is more appropriate to:
1. automatically set object visibility before edit (the easiest)
2. allow changing visibility during edit, or 
3. both.

I think ultimately we need 3. 

Even for situations like editing the sketch. For example when I want to select External geometry that is overlapping with the sketch lines, it would help to temporarily hide the sketch. But it is not possible - sketch visibility seems to be forced on before sketch opens up.

maxwxyz, 2024-03-29:
I would suggest 2.

maxwxyz, 2024-08-25:
@dzid26 by default the eye icon in the tree view is now visible, allowing for a toggle of the visibility of any icon during an active task so option 2 is implemented so I'll close this issue.

closed 2024-08-25 by maxwxyz