FreeCAD/FreeCAD #13224: [Problem] Sketcher_Offset: issues with single line selection + OVP inconsistency

Roy-043, 2024-03-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Create a sketch with a single fully constrained horizontal line.
2. Select it.
3. Call Sketcher_Offset.
4. Check the "Add offset constraint" option.
5. Pick a point to define the distance,
6. Result: the sketch is over-constrained.
7. Delete the proposed constraint.
8. Result: under-constrained: 4 DoFs.

The offset shape has 4 lines (why?) and if you skip step 4 the resulting shape is not stable.

The "Add offset constraint" checkbox is inconsistent if you compare how other OVP tools work. IMO it should be removed. If the user enters a value a constraint should be added automatically.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36596 (Git)
Build type: Release
Branch: main
Hash: 380d452adb5da85f47543a316b6b533f84e829c7
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-30:
@PaddleStroke FYI

PaddleStroke, 2024-04-02:
Offsetting one line is sketchi unfortunately. Occ kernel is handling it bad, it's occ that returns the 4 edges. And in this case I'm not surprised the "offset constraint" is failing.
It is left as an option because it adds a lot of construction geos and sometimes fails.

Roy-043, 2025-06-18:
Still an issue:
```
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.42149 (Git)
Build date: 2025/06/13 08:44:01
Build type: Release
Branch: main
Hash: 9714e2f87a6240d1b2b979eb39c7d34791aafdae
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Free Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```