FreeCAD/FreeCAD #13225: [Problem] V0.22 Sketcher_Rotate does not copy distance constraint attached to edge

Roy-043, 2024-03-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Create a random line.
2. Select the edge and attach a distance constraint (Sketcher_ConstrainDistance).
3. Create copies with Sketcher_Rotate.
4. Result: constraint not copied.
5. Repeat the test but select the endpoints when attaching the constraint.
6. Result: constraint is copied.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36596 (Git)
Build type: Release
Branch: main
Hash: 380d452adb5da85f47543a316b6b533f84e829c7
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-30:
@PaddleStroke FYI

Roy-043, 2025-06-19:
Still an issue.
```
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.42180 (Git)
Build date: 2025/06/16 16:25:59
Build type: Release
Branch: main
Hash: fca41e4003f1680cf604b593a046ac2b1db6ba8a
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Free Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

PaddleStroke, 2025-11-18:
Looking into this