FreeCAD/FreeCAD #13228: Autoconstraints cause conflicting constraints - Please report!

furgo16, 2024-03-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Open attached file
2. Double-click on the sketch to edit it (unrelated: the sketch shows up as a closed perimeter before editing it, although it's open)
3. Click on the "Create line" button
4. Unite the two segments that are open to close the perimeter with a line
5. The operation fails with the  "Autoconstraints cause conflicting constraints - Please report!" error.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36677 (Git) Snap 970
Build type: Release
Branch: main
Hash: 3fadb1f40bdfa4ee1c61acbf12fbe44680723965
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * BIM 2021.12.0
  * Assembly4 0.50.12
  * ExplodedAssembly
```

### Subproject(s) affected?

Sketcher

### Anything else?

File to reproduce: [SmallbuildingGH.zip](https://github.com/FreeCAD/FreeCAD/files/14811195/SmallbuildingGH.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-30:
Do you have the option checked "Auto remove redundant constraints"?

furgo16, 2024-03-30:
> Do you have the option checked "Auto remove redundant constraints"?

@maxwxyz thanks. No, I do not. I've updated the description accordingly.

Enabling the "Auto remove redundant constraints" option results in strange behaviour as well:

Steps to reproduce:

1. Open attached file
1. Double-click on the sketch to edit it (unrelated: the sketch shows up as a closed perimeter before editing it, although it's open)
1. Note: the "Auto constraints" and  "Auto remove redundant constraints" options are set in the Constraints panel
1. Click on the "Create line" button
1. Unite the two segments that are open to close the perimeter with a line
1. The operation fails with the "Autoconstraints cause conflicting constraints - Please report!" error. The line is not drawn
1. Close the sketcher with the "Close" button in the Task panel
1. The line now appears
1. Double-click on the sketch to edit it: the line is now there, unconstrained

maxwxyz, 2024-03-30:
@PaddleStroke it seems that it wants to add coincident and vertical constraints but its already vertical  from the other dimensions:
[image]

closed 2025-09-05 by github-actions[bot]