FreeCAD/FreeCAD #13230: [Feature] Loop counter for preselection so that the zooming of large assemblies is not slowed down

wwmayer, 2024-03-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

https://forum.freecad.org/viewtopic.php?t=86554
https://forum.freecad.org/viewtopic.php?t=86537
https://forum.freecad.org/viewtopic.php?t=85502

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36680 (Git)
Build type: debug
Branch: main
Hash: 82e1091c19269753beab75ee01650d5371021b37
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

Core

### Anything else?

A possible solution could be to add a further option "Auto" besides "On" and "Off" for the pre-selection.

This measures the time span between several rendering cycles. If it exceeds a certain threshold (could also be a user-defined value) then the pre-selection will be switched off automatically. The user should be informed when the internal state changes.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-03-30:
See also https://github.com/FreeCAD/FreeCAD/issues/7214

luzpaz, 2024-08-24:
@PaddleStroke just an FYI to keep on for optimization related to assemblies

maxwxyz, 2024-10-16:
Any news on this issue?