FreeCAD/FreeCAD #13237: Duplicated Shape artifacts when using Part::FeaturePython with App::GroupExtensionPython

morganrallen, 2024-03-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating a `Part::FeaturePython` that uses the `App::GroupExtensionPython`, `Shape` elements inside the `Group` get duplicated on the Shape of the `Part::FeaturePython`.  I've encounter this a couple times and it can vary slightly from one scenario to another. Like sometimes the entire contents of the `Group` get duplicated, but other times a stray `Face` will be created for each child in the group.

In one of the cases I've encounter this, I'm attempting to wrap up a kicadStepUp import in a `PythonFeature`, it works except the contents of the group get duplicated. In the screenshots below if I highlight the main Face of the object on the left, I see `feature_group_bug.PCB.Face101` in the Preselect status line. Highlighting the same face on the object on the right it reads `feature_group_bug.Board_65e7.Board_Geoms_65e7.Pcb_65e7.Face101`

[image]

original discussion here: https://forum.freecad.org/viewtopic.php?t=73214
example file: [feature_group_bug.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/14812957/feature_group_bug.FCStd.zip)

### Full version info

```shell
OS: Arch Linux (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36683 (Git)
Build type: Unknown
Branch: main
Hash: 6fbd8916ab4ae045fdaa4e16a3e44d7393b13e80
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * Render 2024.1.23
  * Plywood_WB
  * Launcher
  * BIM 2021.12.0
  * Curves 0.6.23
  * sheetmetal 0.4.0
  * A2plus 0.4.62
  * Trails 2022.1.0
  * kicadStepUpMod.backup1663197104.2499094
  * SelectorToolbar
  * BIM.backup1699302964.3848543 2021.12.0 (Disabled)
  * fasteners 0.5.1
  * Behave-Dark-Colors 0.1.1
  * kicadStepUpMod 10.23.0
  * Assembly4 0.50.7
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-03-30:
@easyw would you be able to help with this issue ?

morganrallen, 2024-03-31:
yeah, would certainly help where I can. So far I've attempted to find a cause, or workaround from the Python side of things without much success. I've also created a minimal macro to recreate the issue, this should be helpful debugging the issue further.

```
import FreeCAD as App
import Part

def create(obj_name):
    """
    Object creation method
    """

    obj = App.ActiveDocument.addObject('Part::FeaturePython', obj_name)

    box(obj)

    ViewProviderBox(obj.ViewObject)

    App.ActiveDocument.recompute()

    return obj

class box():

    def __init__(self, obj):
        """
        Default constructor
        """

        self.Type = 'box'

        obj.Proxy = self

        obj.addProperty('App::PropertyString', 'Description', 'Base', 'Box description').Description = ""
        obj.addProperty('App::PropertyLength', 'Length', 'Dimensions', 'Box length').Length = 10.0
        obj.addProperty('App::PropertyLength', 'Width', 'Dimensions', 'Box width').Width = '10 mm'
        obj.addProperty('App::PropertyLength', 'Height', 'Dimensions', 'Box height').Height = '1 cm'
        
        obj.addExtension('App::GroupExtensionPython')

    def addObject(self, incoming):
        print(self, incoming.Label)
        for parent in incoming.InList:
            print(self.Object ,parent)
            if (parent != self.Object and incoming in parent.Group):
                print("removing", incoming, "from", parent)
                p_group = parent.Group
                p_group.remove(incoming)
                parent.Group = p_group

        self.Object.Group = self.Object.Group + [ incoming ]

    def execute(self, obj):
        """
        Called on document recompute
        """

        obj.Shape = Part.makeBox(obj.Length, obj.Width, obj.Height)

class ViewProviderBox:

    def __init__(self, obj):
        """
        Set this object to the proxy object of the actual view provider
        """

        obj.Proxy = self
        obj.addExtension('Gui::ViewProviderGroupExtensionPython')

    def attach(self, obj):
        """
        Setup the scene sub-graph of the view provider, this method is mandatory
        """
        return

    def canDropObject(self, obj):
        return True

    def loads(self,state):
        """
        Called during document restore.
        """
        return None

o1 = create('test')
o2 = create('test')
o1.addObject(o2)
o1.Placement.Base.x = -15
```

morganrallen, 2024-04-01:
ok, here's a minimal macro that I think demonstrates the problem. During the actual `FeaturePython` and `ViewProvider` creation parts, nothing is actually added to the `PythonFeature.Shape`. Despite this, when the `Body` is added as a group member, the `PythonFeature` assumes the shape of it's child, effectively duplicating it.

```
import FreeCAD as App
import Part

def create(obj_name):
    obj = App.ActiveDocument.addObject('Part::FeaturePython', obj_name)

    box(obj)

    ViewProviderBox(obj.ViewObject)

    return obj

class box():
    def __init__(self, obj):
        obj.addExtension('App::GroupExtensionPython')

class ViewProviderBox:

    def __init__(self, obj):
        obj.Proxy = self
        obj.addExtension('Gui::ViewProviderGroupExtensionPython')

    def getIcon(self):
        return """
            /* XPM */
            static const char * ViewProviderBox_xpm[] = {
            "16 16 6 1",
            "    c None",
            ".   c #141010",
            "+   c #615BD2",
            "@   c #C39D55",
            "#   c #000000",
            "$   c #57C355",
            "        ........",
            "   ......++..+..",
            "   .@@@@.++..++.",
            "   .@@@@.++..++.",
            "   .@@  .++++++.",
            "  ..@@  .++..++.",
            "###@@@@ .++..++.",
            "##$.@@$#.++++++.",
            "#$#$.$$$........",
            "#$$#######      ",
            "#$$#$$$$$#      ",
            "#$$#$$$$$#      ",
            "#$$#$$$$$#      ",
            " #$#$$$$$#      ",
            "  ##$$$$$#      ",
            "   #######      "};
            """

    def canDropObject(self, obj):
        return True

o1 = create('test')

body = App.ActiveDocument.addObject('PartDesign::Body','Body')
box = App.ActiveDocument.addObject('PartDesign::AdditiveBox','Box')

box.Length = 10
box.Width = 10
box.Height = 10

body.addObject(box)
o1.addObject(body)
o1.Placement.Base.x = 12

App.ActiveDocument.recompute()
```

morganrallen, 2024-04-01:
Couple other things I've noticed, that seem like problematic behavior...

If the Placement on the Feature is changed, it's Shape is moved, but the Body in the Group does not. My feeling for how this should work, is if a Group is moved, all of it's Children should also be moved accordingly. Also in this case the Placement of neither the Feature nor its Shape are updated, which I >guess< hints at a rendering error?

morganrallen, 2024-04-30:
I've been thinking about this issue some more and maybe I've been thinking about it wrong. Could it be the core of this isn't isn't the fact that the Shape is being duplicated, but more that it's not duplicating enough, the styling specifically?

Looking at other parts of FC, a Boolean operation for example, you stop showing the underlying Objects and show the resulting operation. So I wonder if it's better to hide the underlying Object/Shape and reframe this as a question of; should the Group copy styling for it's constituent parts?

luzpaz, 2025-06-23:
bumping for visibility.  
BTW, @morganrallen I enjoyed your KiCon25 talk on KiConnect (https://blog.freecad.org/2025/06/20/kiconnect-at-kicon-new-interoperability-between-freecad-and-kicad/)

morganrallen, 2025-06-23:
Hey thanks! It leans heavily on `PythonFeature`. I've also been meaning to write up some followup thoughts this this, as I've got more experienced with `*Python` like objects, I'm leaning towards this being the correct behavior, but maybe it's missed named and there is a missing `xPython` type. From my point of view this is acting like a `PartDesign::Body`, which would make sense that it would hide the prior Shape and show the new Feature. Think of a `Boolean` in part design, it hides the component features and shows the result of the Boolean.

But what I (and I believe carlopav in the post above) wanted was something like a `App::Part`, something that doesn't have a Shape but can contain multiple other Objects and is movable in 3d space.

There might be some way to achieve this with Extensions, but as of now I see it as a missing Feature type, something like `App::PartPython`

closed 2025-09-07 by github-actions[bot]