FreeCAD/FreeCAD #13239: [Problem] V0.22 Sketcher_Translate: R key use conflicts with R shortcut

Roy-043, 2024-03-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Apparently there is an undocumented(?) standard "R" shortcut that calls the current Sketcher_CompConstrainRadDia tool. This means that using "R" to increase the number of rows for Sketcher_Translate is not possible. Instead the tool is cancelled and the Sketcher_CompConstrainRadDia tool is started.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36596 (Git)
Build type: Release
Branch: main
Hash: 380d452adb5da85f47543a316b6b533f84e829c7
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-06-28:
@PaddleStroke

Roy-043, 2024-06-28:
AFAICT this has been fixed. The R shortcut no longer works.

FEA-eng, 2024-06-29:
> AFAICT this has been fixed.

So the issue might be closed? I can indeed use the R key for the Sketcher Translate rows field.

Roy-043, 2024-07-01:
Indeed this can be closed.

closed 2024-07-01 by Roy-043