FreeCAD/FreeCAD #13240: [Problem] V0.22 Sketcher_Scale and Sketcher_Translate: "Spinbox doesn't exist" warning if OVP disabled

Roy-043, 2024-03-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Both Sketcher_Scale and Sketcher_Translate show a "Spinbox doesn't exist" warning if On-View-Parameters are disabled.

Exact text of the warning:
"inEditableDatumLabel::setSpinboxValue: Spinbox doesn't exist"

Sketcher_Scale:
1. Select a sketch element.
2. Call the tool.
3. Pick the 1st point.
4. Pick the 2nd point.
5. Warning.

Sketcher_Translate:
1. Select a sketch element.
2. Call the tool.
3. Pick the 1st point.
4. Warning.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36596 (Git)
Build type: Release
Branch: main
Hash: 380d452adb5da85f47543a316b6b533f84e829c7
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-06-28:
@PaddleStroke

maxwxyz, 2024-06-29:
@Roy-043 off topic: any reason you remove the 1.0 label on some issues and using the milestone instead? Now we have two labels to consider. If there is any benefit to the milestone I can relabel all which have 1.0 tags

FEA-eng, 2024-06-29:
I would keep the 1.0 label here. This issue is quite important. A ton of warnings appears when moving the cursor with OVP disabled.

Roy-043, 2024-07-01:
Sorry about that. I thought the milestone would be clearer to more contributors. I can put the 1.0 label back if that helps.

FEA-eng, 2024-07-01:
> Sorry about that. I thought the milestone would be clearer to more contributors. I can put the 1.0 label back if that helps.

IMO both are helpful since the label is well-visible and you can sort by it while the milestone has a separate page with progress tracking.

maxwxyz, 2024-07-01:
I'm fine with both, should be consistent. We can discuss it in todays meeting. Tracking progress is good.

closed 2024-07-08 by chennes