FreeCAD/FreeCAD #13243: V0.22dev -  Sketcher constraints toolbar icons dissapear

7th-fret, 2024-03-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Sometimes when i open an existing or create a new sketch, most of the icons of sketcher constraints toolbar dissapears.
If i exit sketcher and reopen the sketch, the icons appears again but in a different order.
Once the problem happens, the order of icons never get back to normal until i reopen FreeCAD

Unfortunately the problem happens randomly, so I can't find a way to reproduce it.

### Full version info

```shell
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36683 (Git)
Build type: Release
Branch: main
Hash: 6fbd8916ab4ae045fdaa4e16a3e44d7393b13e80
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * InventorLoader 1.4.0
  * Curves 0.6.29
  * freecad.gears 1.2.0
  * CurvedShapes 1.0.8
  * sheetmetal 0.4.9
  * fasteners 0.5.15
  * OpticsWorkbench 1.0.12
  * Render 2024.1.23
  * OpenDark 2023.12.17
  * A2plus 0.4.64
```

### Subproject(s) affected?

None

### Anything else?

Normal icons
[image]

Dissapeared icons
[image]

Reappeared icons
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-31:
I cannot reproduce this. Is it disappearing entirely or is the toolbar just moved somewhere else?

7th-fret, 2024-03-31:
> I cannot reproduce this. Is it disappearing entirely or is the toolbar just moved somewhere else?

The toolbar remains visible and at its place, it is the icons of the toolbar that dissapears. As shown in the attached pictures.
As of reproduce, i mentioned it before that it happens radomly and i can't find a way to reproduce.

Roy-043, 2024-03-31:
This has been reported on the Forum as well (and also without a reproducible scenario):
https://forum.freecad.org/viewtopic.php?t=86455

mfraser, 2024-03-31:
I can confirm this has happened to me, but I have been unable to reproduce the steps required to cause it.

maxwxyz, 2024-04-01:
@PaddleStroke do you know this behavior?
@obelisk79 did you experience this while working on the concise toolbars?

obelisk79, 2024-04-01:
I have not experienced this problem yet.

mfraser, 2024-04-01:
Just happened again, but I still can't reproduce it reliably.

luzpaz, 2024-04-01:
@mfraser can you please post full About info, TIA

mfraser, 2024-04-01:
> @mfraser can you please post full About info, TIA
```
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36690 (Git)
Build type: Release
Branch: main
Hash: 7cbb28ca532d652a942e4917a2799c387778ea61
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * QuickMeasure 2022.10.28
  * kicadStepUpMod 11.1.0
  * fasteners 0.5.15
  * Curves 0.6.29
  * ThreadProfile 1.89.0
  * sheetmetal 0.4.10
  * VendorParts
  * lattice2 1.0.0
  * Freecad-Built-in-themes-beta 1.2.2
  * Lithophane
  * ProDarkThemePreferencePack 1.0.0
  * Manipulator 1.5.7
  * DynamicData 2.60.0
  * Defeaturing 1.2.2
  * CurvedShapes 1.0.8
  * Assembly4 0.12.4
  * freecad.gears 1.2.0
  * A2plus 0.4.64
  * fcgear 1.0.0
```

PaddleStroke, 2024-04-02:
Without reproduction steps it is hard to say.
However I have heard before of the toolbars mis-behaving if you change of document while in edit mode. Is that the problem you are referencing to?

mfraser, 2024-04-02:
> Without reproduction steps it is hard to say. However I have heard before of the toolbars mis-behaving if you change of document while in edit mode or something similar I can't remember precisely. Is that the problem you are referencing to?

What do you mean by "change of document"?

mfraser, 2024-04-04:
Is it always the same constraint icons that disappear?

7th-fret, 2024-04-04:
> Is it always the same constraint icons that disappear?

In my case it is always the same icons.

[image]

maxwxyz, 2024-04-05:
might it have to do with the unified option for coincident @PaddleStroke

7th-fret, 2024-04-05:
> might it have to do with the unified option for coincident @PaddleStroke

Maybe, but i have already try to use FreeCAD with unified coincident turned off and the problem persists

mpfusion, 2024-04-05:
I can confirm the behaviour. The same icons occasionally vanish in my sketches as well. However, I can't reproduce it reliably.

maxwxyz, 2024-06-06:
@kadet1090 FYI

depthoffocus, 2024-06-11:
> Is it always the same constraint icons that disappear?

For me it is not, I think -- sometimes I lose just tangent and onwards, sometimes coincident and onwards. 

I haven't managed to make a reproducible test either.

maxwxyz, 2024-08-17:
any news on this issue? It keeps coming up randomly unfortunately.

luzpaz, 2024-08-17:
Added this as a blocker because it's quite intrusive

kadet1090, 2024-08-17:
Without a way to reproduce it is very hard to tackle. I looked at the code and have not found anything useful - not to say that it is impossible to check if this is fixed without reproduce steps. @maxwxyz @luzpaz can you play a little bit and try to come up with steps to reproduce this issue reliably?

kadet1090, 2024-08-17:
So, the problems seems to be in the `void ToolBarManager::setup(ToolBarItem* item, QToolBar* toolbar) const` function: https://github.com/FreeCAD/FreeCAD/blob/8dc1fad1c9d1ba4156b05c3508a85674520ccf24/src/Gui/ToolBarManager.cpp#L775-L812

Actions that should not be present in the toolbar are removed here: https://github.com/FreeCAD/FreeCAD/blob/8dc1fad1c9d1ba4156b05c3508a85674520ccf24/src/Gui/ToolBarManager.cpp#L808-L811. This method is every time that sketch is edited.

Actions are always inserted at the end of the toolbar so, If we assume that:
1. At first all actions that should be present are returned correctly
2. For some random call it returns truncated list of actions
3. The next call action list is returned correctly.

The result would be that missing actions would be inserted at the end - which is exactly what happens.

Strange thing is that we are missing geometric constraints, which are defined here: https://github.com/FreeCAD/FreeCAD/blob/95c514fca3190c7b56c40f2fc4005a7337f64a56/src/Mod/Sketcher/Gui/Workbench.cpp#L521-L526 which is not guarded by any if statement. Even stranger - the "toggle constraint" action is added not removed so it is not like list simply truncated.

This is not possible, so we need to find another explanation. Another reason why it would be deleted if somehow the action was not found in the toolbar even if present on the list. Actions are found by the `data` attritube of `QAction` - that is clearly set. The only explanation is that for this particular set of commands it is somehow overridden. Looking at the code we find the same set of commands in one place: https://github.com/FreeCAD/FreeCAD/blob/95c514fca3190c7b56c40f2fc4005a7337f64a56/src/Mod/Sketcher/Gui/CommandConstraints.cpp#L1271-L1298

If my theory is right (spoiler alert: it is) it should lead us to the reproduction strategy.
1. Open Sketch
2. Open context menu (right click)
3. Close sketch
4. Reopen sketch

Which for me results in 100% of reproduce - it also explains why it is seemingly random. Looking at the code more we can confirm that this is indeed the case: https://github.com/FreeCAD/FreeCAD/blob/95c514fca3190c7b56c40f2fc4005a7337f64a56/src/Gui/Action.cpp#L547-L553

I'll submit PR tomorrow when I figure a good way to make both scenarios work.

obelisk79, 2024-08-17:
Can confirm, steps to reproduce are valid here.

mfraser, 2024-08-18:
I'll also add that if you close and reopen the sketch again the icons reappear.

luzpaz, 2024-08-21:
@mfraser if you have the capability, could you test the fix in #15933 ?

PR #15933 (2024-08-26): Gui: Fix issue with missing icons in sketcher
This boiled down to conflict of using `data` attribute of the QAction in ActionGroup and ToolBarManager. I reworked ActionGroup here to not use the data attribute but this is not a proper solution - but it should be working well enough.

Proper solution would be to leave the data for the specific usecases (like index) and store the command name in some other place. It would however require us to subclass the QAction which is not something that should be done so late in the release cycle.

More information here: https://github.com/FreeCAD/FreeCAD/is [... cut]
closed 2024-08-26 by chennes