FreeCAD/FreeCAD #13247: document or object Named by simple Chinese can't displayed in python console

arthurx70, 2024-03-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

document or object Named by simple Chinese can't displayed in python console, but in message view is displayed well. I try to write Chinese to Json file, it output like" ....   
            "Name": "b__xe5_xba_xa7_001_",
            "Label": "\u5ea7_001","
[image]

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.Unknown
Build type: Release
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Chinese/China (zh_CN)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-31:
can you try if this issue is still in the current weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

luzpaz, 2024-04-01:
Known issue.  
Duplicate of #6858

closed 2024-04-01 by luzpaz

arthurx70, 2024-04-01:
I don't understand, It's mean no one fix it?
I have try [weekly-builds](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds),the problem still exists。

arthurx70, 2024-04-01:
I have try [weekly-builds](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds),the problem still exists。
It's mean I have to build [realthunder](https://github.com/realthunder/FreeCAD/commits?author=realthunder)?

luzpaz, 2024-04-01:
No.. you can use Realthunder's pre-built binaries (https://github.com/realthunder/FreeCAD/releases) but remember his branch (called 'Link') is experimental and independent of FreeCAD main branch