FreeCAD/FreeCAD #13248: [Problem] Invalid SubElementNames

Roy-043, 2024-03-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Newly created Draft_Clones have invalid SubElementNames. Saving and reopening the file fixes this. This is a regression, the problem does not occur in V0.21.

I think this is a problem in Core, not in the Draft Workbench.

1. Create a Part_Box.
2. Make it invisible.
3. Create a Draft_Clone from it.
4. Select a face of the clone.
5. Paste `FreeCADGui.Selection.getSelectionEx("", 0)[0].SubElementNames` in the Python console.
6. Result is something like: `(';Face3;:Hae1,F.Face3',)`. It should be: `('Face3',)`.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36596 (Git)
Build type: Release
Branch: main
Hash: 380d452adb5da85f47543a316b6b533f84e829c7
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-04-01:
> Result is something like: (';Face3;:Hae1,F.Face3',). It should be: ('Face3',).

Looks like some Topo Naming code is active. The method SelectionSingleton::checkSelection() gets as input "Face3" but then sets the sub-name of the `_SelObj` to that cryptic name.

CalligaroV, 2024-04-04:
Hi @Roy-043 and @wwmayer!

I gave a look at this issue and yes, it's related to some modifications introduced for TNP.

For now I found a fix (probably temporary) that should solve both this issue and #13009 .
The full fix is more complicated and needs to import more code from RT fork, I already started working on that.

Regardless for that, I opened #13297 that contains the temporary fix.
Can you try it? On my local build it works.

TIA!

Roy-043, 2024-06-13:
This issue is back. See #14196.

Roy-043, 2024-06-13:
Additional finding:
Highlighting of multiple selected subelements of the clone does not work properly. Only the first subelement is highlighted.

bgbsww, 2024-06-16:
That sounds very familiar ...  #14699 may fix this.

Roy-043, 2024-06-22:
I have to reopen again. This is still an issue. Can somebody please add a unit test for this? Something like this:
```python
def are_subelementnames_ok():
    
    import FreeCAD as App
    import FreeCADGui as Gui
    import Part
    
    doc = App.ActiveDocument
    plane = doc.addObject("Part::Plane", "Plane")
    plane.Length = 10
    plane.Width = 10
    extrude = doc.addObject("Part::Extrusion", "Extrude")
    extrude.Base = plane
    extrude.LengthFwd = 10
    doc.recompute()
    
    Gui.Selection.addSelection("", extrude.Name, "Face2", 5, 0, 5)
    
    return FreeCADGui.Selection.getSelectionEx("", 0)[0].SubElementNames == "('Face2',)"
```

bgbsww, 2024-06-22:
The proposed test is unfortunately not correct.  On a working LS3, the selection ( correctly ) includes an element history path:

```
>>> are_subelementnames_ok()
False
>>> FreeCADGui.Selection.getSelectionEx("", 0)[0].SubElementNames
(';Edge2;:G;XTR;:He13:7,F.Face2',)
>>> 
```

The selection issues should be fixed by correctly handling the full element names ( fix for #14897 on the way).  Is there another example you are seeing?

Roy-043, 2024-06-22:
How can it possibly be that the subelement names are different after saving and reopening? And what does that mean for the TNP solution?

bgbsww, 2024-06-22:
They aren't different.  Or if they are, that's a problem.  They are more complete.  Note that Face2 becomes ;Edge2;:G;XTR;:He13:7,F.Face2   and that is properly processed into Face2 using that last period seperator when required.

Roy-043, 2024-06-22:
Well, sorry to have to say this, but after saving and reopening a file with a Part_Extrude or a Draft_Clone, subelement names are different.

After creating the extrude:
`FreeCADGui.Selection.getSelectionEx("", 0)[0].SubElementNames => (';Edge2;:G;XTR;:H30c:7,F.Face2',)`

After saving and reopening:
`FreeCADGui.Selection.getSelectionEx("", 0)[0].SubElementNames => ('Face2',)`

bgbsww, 2024-06-22:
So, again, I wouldn't describe those as different.  Yes, obviously the text is, but what I'm reading is Part_Extrude and Draft_Clone elements are failing to correctly save and restore their elementMaps.  Which is absolutely true, because there's a massive save-and-restore problem I'm still working to get my arms around.  Biggest bug on my plate.  It's still the same subelement though - if the end part after the '.' matches, then it is the same - and could be regenerated, but that's not what we should do.

Thank-you though for the persistence.  You are correct that there is an error here to be addressed; it just isn't that the names are changing - it's that they are incomplete after a reload with a missing path.

Cases where it goes from Face2 to Face4 ( or ANYTHING;YOU-WANT-HG65123,Face4 ) - that's changing, indicative of an actual toponaming detection failure, and a more severe, though generally exact case specific problem.

bgbsww, 2024-06-24:
#14898 addresses saving and reloading and hopefully the consistency issue here.  #14699 addressed the selection error.

There is a remaining question of whether or not python .SubElementNames should return mapped ( full history ) or unmapped ( short ) names.  I can make arguments in both directions.

Less (python module or addon) code is likely to break (not respect TNP mitigations) returning the mapped names, as it is the equivalent reference that the software expects.  It would be more intuitive and prettier to return only the short names, but it this case we'd need to make sure that every python place that accepts a subelementname is properly converting back to the mapped name.  I think this is well covered, but it could be a problem.

Since LS3 returns them in mapped form, I'm defaulting to doing the same unless an argument other than pure aesthetics shows up.

Is there a remaining bug here to be addressed?

Roy-043, 2024-06-28:
Thanks for explaining. I'll close this one.

PR #13297 (2024-04-05): App/Toponaming: GeoFeature - fixes #13009 - fixes #13248
 * Restored previous logic of GeoFeature::getElementName()
 * Added precompiler directive to use the old logic if FC_USE_TNP_FIX isn't defined, otherwise use the new logic
 * Reworked SketchObjectTest::getElementName() as a consequence of the previous points

fixes #13009
fixes #13248
commit 6779c912 (2024-04-05): App/Toponaming: GeoFeature - fixes #13009 - fixes #13248
 * Restored previous logic of GeoFeature::getElementName()
 * Added precompiler directive to use the old logic if FC_USE_TNP_FIX isn't defined, otherwise use the new logic
 * Reworked SketchObjectTest::getElementName() as a consequence of the previous points
PR #14699 (2024-06-17): Toponaming: fix secondElementSelection and SelectionView
This fixes #14521 the reported bug first seen in selections of CAM model objects where only one selection could be made.

It also cleans up the Selection View to show short element names, and not full history names.

Fix #13248
closed 2024-06-28 by Roy-043