FreeCAD/FreeCAD #13249: Flattened SVG export hangs FreeCAD

Fojtik, 2024-03-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I choose "Export frattened SVG", FreeCAD hangs and even after hour it is irresponsitive.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36526 (Git)
Build type: Release
Branch: main
Hash: 2ebbd83268ffabf41ceaad802714a4ffd9214437
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Czech/Czech Republic (cs_CZ)
```

### Subproject(s) affected?

None

### Anything else?

[RabbitSlice.zip](https://github.com/FreeCAD/FreeCAD/files/14817001/RabbitSlice.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-03-31:
There are some issues with the model. You have sliced a shell (not a solid) created from a mesh. You are probably creating an SVG from an object with over 5800 faces. It seems unlikely you want that.

1. Please follow the instructions you can find in the Wiki: https://wiki.freecad.org/Part_ShapeFromMesh.
2. Don't skip the [Part_RefineShape](https://wiki.freecad.org/Part_RefineShape) step.
3. Select the top face of the refined shape.
4. Call [PartDesign_SubShapeBinder](https://wiki.freecad.org/PartDesign_SubShapeBinder).
5. Export the binder to SVG. This will only take a few seconds.
[image]

Fojtik, 2024-03-31:
Thank you very much for help. But still the FreeCAD should show some progress bar and give user a chance to break operation. And not silly hang for unknown amount of hours or even forewer.

Fojtik, 2024-04-01:
I has been upgraded to the yesterday's weekly build but no way, still hangs.
I have hidden everything else
[RabbitSlice.zip](https://github.com/FreeCAD/FreeCAD/files/14820423/RabbitSlice.zip)
and only binder is visible.

Roy-043, 2024-04-01:
You have misunderstood my instructions. Specifically you have not created the binder from the top face only. In your model the binder has over 1700 faces. In the attached model it has only one face.
[RabbitSlice-roy.zip](https://github.com/FreeCAD/FreeCAD/files/14822046/RabbitSlice-roy.zip)

Fojtik, 2024-04-01:
Thanks again.

Anyway, after 2 hours the rabbit SVG gets exported. I do not understand why FreeCAD solves 1700 objects so long. Graphics cards are capable to solve milions of triangles per second.

You are right, selection of top layer shrink export time to seconds.

-----------------------------------------
I still think that this is a bug - not to inform user that there is executed some extra long calculation, that cannot be interrupted from some progress bar.

Fojtik, 2025-06-19:
Exporting object "ExplodedSlice" to flattened SVG still takes extraordirary too much time. It is also interesting 20% CPU load that means that only one core is overloaded. Looks like FreeCAD hangs, no way to stop operation or see percentager counter.

The situation with a new version is even worse. The FreeCAD asks to reindex, but it falls to extraordinary long uninteruptable operation.

Please note that problems does not disappear as time goes on.

FEA-eng, 2025-08-20:
Even opening this .FCStd file hangs FreeCAD on a decent PC but I guess it's due to unnecessarily large number of faces.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/20 00:00:55
Build type: Release
Branch: main
Hash: 5e4fe3a3c3af46be530969d323368fee8bfcba3b
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

Fojtik, 2025-09-20:
Please keep in mind that the main problem is that FreeCAD becomes hanged for several hours! The user cannot stop operation from GUI. No progress bar. FreeCAD looks like it has been crashed.

**This problem still persists in FreeCAD 1.0.2.**

FEA-eng, 2025-11-20:
@Roy-043 Do you think it should be closed? If anything, I guess only this part is an actual issue here:

> I still think that this is a bug - not to inform user that there is executed some extra long calculation, that cannot be interrupted from some progress bar.

closed 2025-11-20 by Roy-043