FreeCAD/FreeCAD #13252: Dangling references to temporaries

NomAnor, 2024-03-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

GCC 13.2.1 show multiple warnings about dangling references to temporaries. They all look similar:

```
../../src/Mod/CAM/App/FeaturePathShape.cpp: In member function ‘virtual App::DocumentObjectExecReturn* Path::FeatureShape::execute()’:
../../src/Mod/CAM/App/FeaturePathShape.cpp:68:29: warning: possibly dangling reference to a temporary [-Wdangling-reference]
   68 |         const TopoDS_Shape &shape = static_cast<Part::Feature*>(*it)->Shape.getShape().getShape();
      |                             ^~~~~
../../src/Mod/CAM/App/FeaturePathShape.cpp:68:96: note: the temporary was destroyed at the end of the full expression ‘Part::PropertyPartShape::getShape() const().Part::TopoShape::getShape()’
   68 |         const TopoDS_Shape &shape = static_cast<Part::Feature*>(*it)->Shape.getShape().getShape();
      |                                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
```

```
../../src/Mod/TechDraw/App/FeatureProjection.cpp: In member function ‘virtual App::DocumentObjectExecReturn* TechDraw::FeatureProjection::execute()’:
../../src/Mod/TechDraw/App/FeatureProjection.cpp:72:25: warning: possibly dangling reference to a temporary [-Wdangling-reference]
   72 |     const TopoDS_Shape& shape = static_cast<Part::Feature*>(link)->Shape.getShape().getShape();
      |                         ^~~~~
../../src/Mod/TechDraw/App/FeatureProjection.cpp:72:93: note: the temporary was destroyed at the end of the full expression ‘Part::PropertyPartShape::getShape() const().Part::TopoShape::getShape()’
   72 |     const TopoDS_Shape& shape = static_cast<Part::Feature*>(link)->Shape.getShape().getShape();
      |                                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
```

```
../../src/Mod/Fem/Gui/TaskFemConstraintFluidBoundary.cpp: In constructor ‘FemGui::TaskFemConstraintFluidBoundary::TaskFemConstraintFluidBoundary(FemGui::ViewProviderFemConstraintFluidBoundary*, QWidget*)’:
../../src/Mod/Fem/Gui/TaskFemConstraintFluidBoundary.cpp:245:37: warning: possibly dangling reference to a temporary [-Wdangling-reference]
  245 |                 const TopoDS_Shape& pShape = pcPart->Shape.getShape().getShape();
      |                                     ^~~~~~
../../src/Mod/Fem/Gui/TaskFemConstraintFluidBoundary.cpp:245:79: note: the temporary was destroyed at the end of the full expression ‘Part::PropertyPartShape::getShape() const().Part::TopoShape::getShape()’
  245 |                 const TopoDS_Shape& pShape = pcPart->Shape.getShape().getShape();
      |                                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
```

### Full version info

```shell
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36695 (Git)
Build type: Debug
Branch: pd-pattern-ref
Hash: 6c0865a7b974cd8100585a42a9b2d21643be871f
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.3.0, OCC 7.3.0
Locale: German/Germany (de_DE)
Installed mods: 
  * Assembly3 0.12.2
  * fasteners 0.5.15
  * DynamicData 2.60.0
```

### Subproject(s) affected?

CAM/Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-04-01:
In the past the code was correct bu this has changed with 2adff99c146 where the const reference has been replaced with a tmp. object

luzpaz, 2024-04-01:
Heads up @bgbsww :point_up_2:

bgbsww, 2024-04-01:
Thanks all, that was clearly awful.  I've rewritten it and I'm just running some more tests both without and with TNP in play before I take the fix out of draft.

closed 2024-04-01 by chennes