FreeCAD/FreeCAD #13266: Sketcher: no snapping to quadrant points

FEA-eng, 2024-04-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Snapping to quadrant points of circles/arcs/ellipses could be useful, possibly with a horizontal/vertical auto constraint like in SolidWorks:

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36391 (Git)
Build type: Release
Branch: main
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-04-02:
This issue is more generic: #11414.

FEA-eng, 2024-04-02:
> This issue is more generic: #11414.

Yeah, but we decided to split it into multiple issues: https://github.com/FreeCAD/FreeCAD/issues/6240#issuecomment-1987923650 which makes sense because the snaps are implemented independently.

kaiwas, 2024-04-03:
> Yeah,

Unfortunately, developers do not consider the sketcher as a “direct” drawing tool. (I would really like this).
Although there are similarities between the bindings and constraints discussed here, each option serves a different purpose.
I hope that someday developers will be able to look at sketcher from a different angle and see that it could be a great drawing tool. Parametric "AutoCAD" would be very cool!

FEA-eng, 2024-04-03:
> Parametric "AutoCAD" would be very cool!

To be honest, I choose sketchers in parametric 3D CAD software (commercial or FreeCAD) when I want to design some 2D shape, check geometrical relationships or sort of kinematics. AutoCAD is good for large 2D drawings in architecture or mechanical engineering, for instance. Here the Draft wb comes as an alternative and has most snaps already available. So it all depends on the purpose. I doubt the Sketcher will ever serve as a tool for larger drawings but features like snaps can make the workflow much more enjoyable even for small drawings. Fortunately, there's a tool that lets you use sketches in Draft and vice versa.